carla-simulator/scenario_runner #795: [OpenSCENARIO] - Using Static Asset as Entityref for a Trigger

LucasFKobernic, 2021-09-08:
**Describe our question or idea**
Hi everyone, 

I'm trying to create a StartTrigger with the condition "RelativeDistanceCondition" between a walker and a static object. 

Until now i have placed the static object defining it as "MiscObject" under "Entities".

It is possible to use directly a asset that already is in the world?  If yes, how should i refer to this asset in the OpenScenario file? 

**Desktop (please complete the following information):**
 - OS: [Windows 10]
 - CARLA Version [0.9.11]
 - Python version [3.8]
 - Version [0.9.11]

fabianoboril, 2021-09-10:
Hi,
just for clarification: You would like to use an object that is already present in the CARLA world? I don't think that this is possible. Or do you want to spawn a specific object?

LucasFKobernic, 2021-09-10:
I want to use a object that is already in CARLA Map.

fabianoboril, 2021-09-10:
That is a challenge... First, we would need to stretch the XOSC definition a bit, which I guess is fine. Second, we would need to get the object through CARLA's PythonAPI. The only way I see is to query all bounding boxes in the world (get_level_bbs). However, this does not provide unique identifiers. So the question would be, is the object you are looking for unique in the scene? If you would need to differentiate among multiple "buildings", you could directly use a Position-Action instead of an Object-Action, right?

FBanani, 2022-08-03:
> 

hi what do you mean by position action I have multiple signs which are around a specific space in Carla world, I used traffic_maneger.py to create other vehicles and set an autopilot mode to them but they crash to the stationary objects on the road. I do not know, what I should do to improve autopilot mode for other vehicle that do not crash with objects in the road, do you have any other idea?