carla-simulator/scenario_runner #798: Scenario Runner: unable to attach actor: parent actor not found

Bedoubleu, 2021-09-14:
Hello Guys,

I'm pretty new to CARLA and Scenario Runner and I hope you can help me.
I've set up my own scenario on a custom map I also created myself. The scenario itself is an ego vehicle controlled with the WaypointFollower and a Walker who crosses a pedestrian crosswalk. When the walker recognises the ego vehicle, the walker should stop to prevent a collision with the ego vehicle. Test Criteria in my scenario is the CollisionTest.
In my .xml config File I created several types of the scenario with different speeds for the walker as well as the ego vehicle which I want to run one after another.

Now to the problem:
If I run the scenario with: _python scenario_runner.py  --scenario group:PedCrossing --reloadWorld --output_
everything works fine with the first two scenarios. "_-All scenario tests were passed successfully!_" is the output I get.

When it comes to the third scenario i get the following Error message: 
[image]

As far as I understand, the collision sensor is the actor that should be connected to the parent, which is the ego vehicle. Please correct me if I'm wrong.
I don't get why it works for the first two scenarios and then suddenly throws the error.

At the moment I'm really clueless how to solve this problem. I hope you can help me to solve this issue.

 - OS: Ubuntu 20.04
 - CARLA Version: 0.9.11
 - Scenario Runner Version: 0.9.11
 - Python Version: 3.8.5

fabianoboril, 2021-09-21:
@glopezdiest Can you look into this?

glopezdiest, 2021-09-21:
That generally tends to happen when you try to attach a sensor to an actor that isn't there, meaning there might have been an error when spawning the actor. Can you make sure that the actor is there before trying to attach the sensor? Probably your initial position makes the actor collide with something (normally the ground). Could you check that?

Bedoubleu, 2021-09-21:
First of all thanks for the reply. 

I tried setting the _criteria_enable=False_ in my "def __init__" to see what is happening. When I do this the first two scenarios work fine again and for the third I get the following output: _srunner.scenariomanager.carla_data_provider.get_location: Actor(id=191, type=vehicle.lincoln.mkz2017) not found!_
What's interesting about this, is that the vehicle is spawned on the map though.

I tried to debug my scenario to find out where the problem is. It seems the scenario_manager.py file calls "self.cleanup()" in the function "def run_scenario(self)" which then calls the "CarlaDataProvider.cleanup()"
[image]
In there I added two print() commands to see if this batch is executed.

In the screenshot below you can see that it is only executed for the first scenario. For me it seems that the ego vehicle gets deleted and not set again for the next scenario, even though it is spawned on the map. But I'm not a 100% sure if I am on the right track here. 
[image]

glopezdiest, 2021-09-23:
That's interesting. From what you told me, you are having the [*The scenario cannot be loaded*](https://github.com/carla-simulator/scenario_runner/blob/master/scenario_runner.py#L398) message. Could you check whether or not the ego vehicle has been properly assigned. [This function](https://github.com/carla-simulator/scenario_runner/blob/master/scenario_runner.py#L379) handles the spawn of the ego, can you check that *self.ego_vehicles* is not empty, None, or something similar? If something happened with the ego, and it is your only actor, the lack of the cleanup messages could make sense.

Bedoubleu, 2021-09-23:
Yes exactly, _The scenario cannot be loaded_ pops up when scenario 3 tries to start.
I checked _self.ego_vehicles_ and it is not empty. But because of the fact that the ego vehicle doesn't get destroyed after scenario 2, I've got two ego_vehicles for scenario 3. as shown in the screenshot below.
[image]

glopezdiest, 2021-09-27:
Okay, so for some reason, the SR cleanup isn't being triggered during the second scenario, which breaks the third. Hmmm, I'll check see if I can reproduce it and report back. In the meantime, I suggest you manually check all cleanup calls and see whether or not they are being called

glopezdiest, 2021-09-27:
Apparently, it was a bug we introduced after some changes to how the cleanup worked. The (one line) PR above should fix it. Please give it a try, see if it works for you

Bedoubleu, 2021-09-27:
I added the line and now it is working perfectly fine!
Thank you guys. Appreciate your help!

closed 2021-09-27 by fabianoboril

FADEAWAY114, 2023-06-13:
> I added the line and now it is working perfectly fine! Thank you guys. Appreciate your help!

Hello, I also got the same error, which line of code was added?

PR #812 (2021-09-27): Fixed repetition crash
### Description

The *finished* attribute was added to avoid multiple cleanups of the simulation. However, this parameter wasn't being reset on repetitions, causing them to crash if 3 or more were run

Fixes #798

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 3.6
  * **Unreal Engine version(s):** 4.26
  * **CARLA version:** 0.9.12

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/812)