carla-simulator/scenario_runner #799: Unable to run scenarios using scenario runner or open scenario

werewolfdev, 2021-09-15:
Trying to run the script python3 scenario_runner.py --scenario FollowLeadingVehicle_1.
Carla world is running

Traceback (most recent call last):
  File "scenario_runner.py", line 35, in <module>
    from srunner.scenarioconfigs.openscenario_configuration import OpenScenarioConfiguration
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/scenarioconfigs/openscenario_configuration.py", line 16, in <module>
    import xmlschema
  File "/home/schemba/.local/lib/python3.7/site-packages/xmlschema/__init__.py", line 15, in <module>
    from .resources import (
  File "/home/schemba/.local/lib/python3.7/site-packages/xmlschema/resources.py", line 14, in <module>
    from elementpath import iter_select, Selector, XPath1Parser
  File "/home/schemba/.local/lib/python3.7/site-packages/elementpath/__init__.py", line 25, in <module>
    from .xpath_context import XPathContext, XPathSchemaContext
  File "/home/schemba/.local/lib/python3.7/site-packages/elementpath/xpath_context.py", line 15, in <module>
    from .xpath_nodes import AttributeNode, TypedAttribute, TypedElement, is_etree_element, \
  File "/home/schemba/.local/lib/python3.7/site-packages/elementpath/xpath_nodes.py", line 19, in <module>
    from .datatypes import ncname_validator
ImportError: cannot import name 'ncname_validator' from 'elementpath.datatypes' (/home/schemba/.local/lib/python3.7/site-packages/elementpath/datatypes/__init__.py)

**Desktop (please complete the following information):**
 - OS: [Ubuntu 18.04]
 - CARLA Version CARLA 0.9.11]
 - Python version [3.7]
 - Version [0.9.11]
Any inputs would be really helpful

fabianoboril, 2021-09-15:
Which version of xmlschema are you using? The requirements specify xmlschema==1.0.18.

werewolfdev, 2021-09-15:
Hi @fabianoboril  Iam using the same

fabianoboril, 2021-09-15:
Can you check which version of elementpath is installed? In my case, it is 1.3.3.

werewolfdev, 2021-09-16:
Hi @fabianoboril . I used python 2.7 with its egg and scenario is getting prepared but cant open manual_control.py now 
Traceback (most recent call last):
  File "manual_control.py", line 58, in <module>
    from examples.manual_control import (World, HUD, KeyboardControl, CameraManager,
ImportError: No module named examples.manual_control
 Thsi si the error that i receive

werewolfdev, 2021-09-16:
I  had to add an init.py file at ~/PythonAPI/examples folder. Can you kindly mentoin it in docs or include in script

closed 2021-09-16 by werewolfdev