carla-simulator/scenario_runner #800: ImportError: No module named examples.manual_control

werewolfdev, 2021-09-16:
Hi @fabianoboril . I used python 2.7 with its egg and scenario is getting prepared but cant run manual_control.py from scenario runner folder now

This is the error that i receive
Traceback (most recent call last):
File "manual_control.py", line 58, in
from examples.manual_control import (World, HUD, KeyboardControl, CameraManager,
ImportError: No module named examples.manual_control

My python path is as follows: 

export PYTHONPATH=$PYTHONPATH:/home/schemba/carla/PythonAPI/carla/dist/carla-0.9.11-py2.7-linux-x86_64.egg:/home/schemba/carla/PythonAPI/carla/agents:/home/schemba/carla/PythonAPI/carla:/home/schemba/carla/PythonAPI:/home/schemba/carla/scenario_runner-0.9.11

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.11]
 - Python version [e.g. 2.7]
 - Version [e.g. 0.9.11]

werewolfdev, 2021-09-16:
I  had to add an init.py file at ~/PythonAPI/examples folder. Can you kindly mentoin it in docs or include in script.

closed 2021-09-16 by werewolfdev

LucasFKobernic, 2022-02-14:
Hi @werewolfdev 

I#M facing the same issue on Windows with scenario runner 0.9.12. Which init.py do u mean?

werewolfdev, 2022-02-14:
Hi @LucasFKobernic .I added an empty document whose name was __init__.py  into the examples folder as seen from another persons issue comments and it worked.