carla-simulator/scenario_runner #801: Controlling Ego Vehicle openSCENARIO

werewolfdev, 2021-09-20:
Hey , I am using openSCENARIO for creating scenarios with CARLA and would like to control the ego vehicle in autopilot , provide a constant velocity or particular dynamics and assign a route to the ego vehicle

I have provided vehicle.set_autopilot(True) in manual_control.py and used the function of enable constant velocity as well.
I would like to know if we can assign a route to the ego vehicle while in autopilot mode . Can you kindly let me know fi there is any support for this.

I couldnot find the simple_vehicle_control in github similar to some other friends who mentioned the same in some other ticket.Thanks in advance

 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.11]
 - Python version [e.g. 2.7]
 - Version [e.g. 0.9.11]

werewolfdev, 2021-09-20:
I could find a simple_vehicle_control python script in the atomics folder .But Iam uncertain of how to use it with openSCENARIO

fabianoboril, 2021-09-20:
Hi,
you can find an example here:
https://github.com/carla-simulator/scenario_runner/blob/05a811a6248c8033e4c76185ff40ae5eccad7e69/srunner/examples/OscControllerExample.xosc#L77
Just replace external_control with simple_vehicle_control

werewolfdev, 2021-09-20:
Hi, Thanks for the quick response. So I replace the control type with simple_vehicle_control and then for providing a route for my ego vehicle how can I go forth doing it. Sorry I couldnt understand it still

werewolfdev, 2021-09-20:
@fabianoboril  Hi, Thanks for the quick response. So I replace the control type with simple_vehicle_control and then for providing a route for my ego vehicle how can I go forth doing it. Sorry I couldnt understand it still

fabianoboril, 2021-09-20:
You provide the route through a Route Maneuver, e.g.

`             <Action name="TakingPrio1Drive2">
                <PrivateAction>
                  <RoutingAction>
                    <AssignRouteAction>
                      <Route name="TurnRight" closed="false">
                        <Waypoint routeStrategy="fastest">
                          <Position>
                            <WorldPosition x="88" y="300" z="0" h="0"/>
                          </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="fastest">
                          <Position>
                            <WorldPosition x="88" y="322" z="0" h="0"/>
                          </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="fastest">
                          <Position>
                            <WorldPosition x="87" y="328" z="0" h="0"/>
                          </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="fastest">
                          <Position>
                            <WorldPosition x="82" y="330" z="0" h="0"/>
                          </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="fastest">
                          <Position>
                            <WorldPosition x="70" y="330" z="0" h="0"/>
                          </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="fastest">
                          <Position>
                            <WorldPosition x="68" y="330" z="0" h="0"/>
                          </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="fastest">
                          <Position>
                            <WorldPosition x="65" y="330" z="0" h="0"/>
                          </Position>
                        </Waypoint>
                      </Route>
                    </AssignRouteAction>
                  </RoutingAction>
                </PrivateAction>
              </Action>
`

werewolfdev, 2021-09-20:
Hi , Will try this out and update here soon

werewolfdev, 2021-09-21:
Preparing scenario: LaneChangeSimple
Warning: Maneuvergroup ManeuverSequence does not use reference actors!
ScenarioManager: Running scenario OpenScenario
Traceback (most recent call last):
  File "scenario_runner.py", line 405, in _load_and_run_scenario
    self.manager.run_scenario()
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/scenariomanager/scenario_manager.py", line 133, in run_scenario
    self._tick_scenario(timestamp)
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/scenariomanager/scenario_manager.py", line 174, in _tick_scenario
    self.scenario_tree.tick_once()
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/behaviour.py", line 158, in tick_once
    for unused in self.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/behaviour.py", line 247, in tick
    self.initialise()
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py", line 746, in initialise
    if not actor_dict or not self._actor.id in actor_dict:
AttributeError: 'NoneType' object has no attribute 'id'
'NoneType' object has no attribute 'id'
Resetting ego-vehicle!
Destroying ego vehicle 207
ERROR: failed to destroy actor 207 : unable to destroy actor: not found 
No more scenarios .... Exiting

After running the xosc file, the ego vehicle just starts to move and the window closes with these error messages. Can you kindly let me know what this is about. Thanks in advance.

werewolfdev, 2021-09-21:
@fabianoboril Preparing scenario: LaneChangeSimple
Warning: Maneuvergroup ManeuverSequence does not use reference actors!
ScenarioManager: Running scenario OpenScenario
Traceback (most recent call last):
  File "scenario_runner.py", line 405, in _load_and_run_scenario
    self.manager.run_scenario()
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/scenariomanager/scenario_manager.py", line 133, in run_scenario
    self._tick_scenario(timestamp)
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/scenariomanager/scenario_manager.py", line 174, in _tick_scenario
    self.scenario_tree.tick_once()
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/behaviour.py", line 158, in tick_once
    for unused in self.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/schemba/.local/lib/python2.7/site-packages/py_trees/behaviour.py", line 247, in tick
    self.initialise()
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py", line 746, in initialise
    if not actor_dict or not self._actor.id in actor_dict:
AttributeError: 'NoneType' object has no attribute 'id'
'NoneType' object has no attribute 'id'
Resetting ego-vehicle!
Destroying ego vehicle 207
ERROR: failed to destroy actor 207 : unable to destroy actor: not found 
No more scenarios .... Exiting

After running the xosc file, the ego vehicle just starts to move and the window closes with these error messages. Can you kindly let me know what this is about. Thanks in advance.

fabianoboril, 2021-09-21:
Can you share the scenario file?

werewolfdev, 2021-09-21:
Yes sure @fabianoboril

werewolfdev, 2021-09-21:
<?xml version="1.0"?>
<OpenSCENARIO>
  <FileHeader revMajor="1" revMinor="0" date="2020-03-24T12:00:00" description="CARLA:AusweichAssistant_01" author=""/>
  <ParameterDeclarations/>
  <CatalogLocations/>
  <RoadNetwork>
    <LogicFile filepath="Town04"/>
    <SceneGraphFile filepath=""/>
  </RoadNetwork>
  <Entities>
    <ScenarioObject name="hero">
      <Vehicle name="vehicle.tesla.model3" vehicleCategory="car">
        <ParameterDeclarations/>
        <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties>
          <Property name="type" value="ego_vehicle"/>
        </Properties>
      </Vehicle>
    </ScenarioObject>
    <ScenarioObject name="standing">
      <Vehicle name="vehicle.volkswagen.t2" vehicleCategory="car">
        <ParameterDeclarations/>
        <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties>
          <Property name="type" value="simulation"/>
        </Properties>
      </Vehicle>
    </ScenarioObject>
  </Entities>
  <Storyboard>
    <Init>
      <Actions>
        <GlobalAction>
          <EnvironmentAction>
            <Environment name="Environment1">
              <TimeOfDay animation="false" dateTime="2019-06-25T12:00:00"/>
              <Weather cloudState="free">
                <Sun intensity="0.85" azimuth="0" elevation="1.31"/>
                <Fog visualRange="100000.0"/>
                <Precipitation precipitationType="dry" intensity="0.0"/>
              </Weather>
              <RoadCondition frictionScaleFactor="1.0"/>
            </Environment>
          </EnvironmentAction>
        </GlobalAction>
        <Private entityRef="hero">
          <PrivateAction>
            <TeleportAction>
              <Position>
                <WorldPosition x="-9.4" y="-152.8" z="0.5" h="1.57079632679"/>
              </Position>
            </TeleportAction>
          </PrivateAction>
          <PrivateAction>
[... 137 lines cut ...]
        <Condition name="criteria_OnSidewalkTest" delay="0" conditionEdge="rising">
          <ByValueCondition>
            <ParameterCondition parameterRef="" value="" rule="lessThan"/>
          </ByValueCondition>
        </Condition>
        <Condition name="criteria_KeepLaneTest" delay="0" conditionEdge="rising">
          <ByValueCondition>
            <ParameterCondition parameterRef="" value="" rule="lessThan"/>
          </ByValueCondition>
        </Condition>
        <Condition name="criteria_CollisionTest" delay="0" conditionEdge="rising">
          <ByValueCondition>
            <ParameterCondition parameterRef="" value="" rule="lessThan"/>
          </ByValueCondition>
        </Condition>
        <Condition name="criteria_DrivenDistanceTest" delay="0" conditionEdge="rising">
          <ByValueCondition>
            <ParameterCondition parameterRef="distance_success" value="100" rule="lessThan"/>
          </ByValueCondition>
        </Condition>
      </ConditionGroup>
    </StopTrigger>
  </Storyboard>
</OpenSCENARIO>
And I have provided vehilce.set_autopilot(True) in manual_control.py

werewolfdev, 2021-09-21:
<?xml version="1.0"?>
<OpenSCENARIO>
  <FileHeader revMajor="1" revMinor="0" date="2020-03-24T12:00:00" description="CARLA:AusweichAssistant_01" author=""/>
  <ParameterDeclarations/>
  <CatalogLocations/>
  <RoadNetwork>
    <LogicFile filepath="Town04"/>
    <SceneGraphFile filepath=""/>
  </RoadNetwork>
  <Entities>
    <ScenarioObject name="hero">
      <Vehicle name="vehicle.tesla.model3" vehicleCategory="car">
        <ParameterDeclarations/>
        <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties>
          <Property name="type" value="ego_vehicle"/>
        </Properties>
      </Vehicle>
    </ScenarioObject>
    <ScenarioObject name="standing">
      <Vehicle name="vehicle.volkswagen.t2" vehicleCategory="car">
        <ParameterDeclarations/>
        <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties>
          <Property name="type" value="simulation"/>
        </Properties>
      </Vehicle>
    </ScenarioObject>
  </Entities>
  <Storyboard>
    <Init>
      <Actions>
        <GlobalAction>
          <EnvironmentAction>
            <Environment name="Environment1">
              <TimeOfDay animation="false" dateTime="2019-06-25T12:00:00"/>
              <Weather cloudState="free">
                <Sun intensity="0.85" azimuth="0" elevation="1.31"/>
                <Fog visualRange="100000.0"/>
                <Precipitation precipitationType="dry" intensity="0.0"/>
              </Weather>
              <RoadCondition frictionScaleFactor="1.0"/>
            </Environment>
          </EnvironmentAction>
        </GlobalAction>
        <Private entityRef="hero">
          <PrivateAction>
            <TeleportAction>
              <Position>
                <WorldPosition x="-9.4" y="-152.8" z="0.5" h="1.57079632679"/>
              </Position>
            </TeleportAction>
          </PrivateAction>
          <PrivateAction>
[... 136 lines cut ...]
        </Condition>
        <Condition name="criteria_OnSidewalkTest" delay="0" conditionEdge="rising">
          <ByValueCondition>
            <ParameterCondition parameterRef="" value="" rule="lessThan"/>
          </ByValueCondition>
        </Condition>
        <Condition name="criteria_KeepLaneTest" delay="0" conditionEdge="rising">
          <ByValueCondition>
            <ParameterCondition parameterRef="" value="" rule="lessThan"/>
          </ByValueCondition>
        </Condition>
        <Condition name="criteria_CollisionTest" delay="0" conditionEdge="rising">
          <ByValueCondition>
            <ParameterCondition parameterRef="" value="" rule="lessThan"/>
          </ByValueCondition>
        </Condition>
        <Condition name="criteria_DrivenDistanceTest" delay="0" conditionEdge="rising">
          <ByValueCondition>
            <ParameterCondition parameterRef="distance_success" value="100" rule="lessThan"/>
          </ByValueCondition>
        </Condition>
      </ConditionGroup>
    </StopTrigger>
  </Storyboard>
</OpenSCENARIO>

werewolfdev, 2021-09-21:
[AusweichAssistant_01.txt](https://github.com/carla-simulator/scenario_runner/files/7200873/AusweichAssistant_01.txt)

werewolfdev, 2021-09-21:
@fabianoboril I have attached the xosc  file

fabianoboril, 2021-09-21:
You don't need manual control anymore, as the ego vehicle is now controlled through simple_vehicle_control. If you like, you can add a camera like this:

>                                 <Controller name="HeroAgent">
>                                     <Properties>
>                                         <Property name="module" value="simple_vehicle_control"/>
>                                         <Property name="attach_camera" value="true"/>
>                                     </Properties>
>                                 </Controller>

In our scenario, there is one mistake:

>         <ManeuverGroup maximumExecutionCount="1" name="ManeuverSequence">
>           <Actors selectTriggeringEntities="false">
>             <EntityRef entityRef="ego_vehicle"/>
>           </Actors>

Instead of ego_vehicle, this has to be hero. 

Have you seen the graphical scenario creation tool? I think this will be helpful, to avoid such mistakes.

werewolfdev, 2021-09-21:
Hi I shoukd run simple_vehicle_control.py now for running the scenarios you mean to say , after running the xosc file

werewolfdev, 2021-09-21:
Reagrding the graphical scenario creation tool, do you mean QGIS scenario editor. In other cases I havent seen

werewolfdev, 2021-09-21:
@fabianoboril I received a small window with the car stagnant and not moving. 2 questions about this:

a) The ego vehicle was expected to follow that route. Am i missing anything for carrying out this manuever
b )Is there are a way to increase the window size

werewolfdev, 2021-09-21:
@fabianoboril I received a small window with the car stagnant and not moving. 2 questions about this:

a) The ego vehicle is moving but cannot find the stationary target
 b)Is there are a way to increase the window size
c)The coordinates x and y displayed on the window doesnt match the ones given in the xosc for teleport action

werewolfdev, 2021-09-21:
Hi @fabianoboril  Thanks for that. But I couldnot find the stationary object which I have coded in the above txt .Also other xosc scenarios which I have created just seems to be run but without displaying any output window. Do you have any inputs on these questions too. I have actually already added camera as you mentioned while defining the controller

werewolfdev, 2021-09-21:
ABout visualizer.py, Iam suing SR 0.9.11 and has no visualizer python script inside actor controls folder . Any comments on that too would be helpful

fabianoboril, 2021-09-21:
Can you update to the 0.9.11 branch (not tag)? This will most likely solve many of your issues.

werewolfdev, 2021-09-21:
Hi @fabianoboril No module named simple watch dog timer import error pops up when trying to run xosc files now

werewolfdev, 2021-09-21:
Hi @fabianoboril 
Traceback (most recent call last):
  File "scenario_runner.py", line 37, in <module>
    from srunner.scenariomanager.scenario_manager import ScenarioManager
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/scenariomanager/scenario_manager.py", line 23, in <module>
    from srunner.scenariomanager.watchdog import Watchdog
  File "/home/schemba/carla/scenario_runner-0.9.11/srunner/scenariomanager/watchdog.py", line 14, in <module>
    import simple_watchdog_timer as swt
  File "/home/schemba/.local/lib/python2.7/site-packages/simple_watchdog_timer.py", line 11
    def __init__(self, callback, check_interval_sec: float = 0.01, trigger_delta_sec: float = 1, identifier=None):
                                                   ^
SyntaxError: invalid syntax
 Received this error when trying to run a xosc example from the 0.9.11 branch

werewolfdev, 2021-09-22:
Hi @fabianoboril  Do you ahve any insights on this

NaveedBhuiyan, 2023-02-17:
> 

incase of adding the camera would adding that single line suffice? Is there any way to save the images that come from the camera sensors by defining in the xosc file ?

arjunsk07, 2023-03-06:
@werewolfdev Were you able to run the xosc file successfully?

silakuzu, 2025-04-10:
Hello, could you control the ego vehicle at the end? @arjunsk07 @werewolfdev