carla-simulator/scenario_runner #802: openSCENARIO Editor

werewolfdev, 2021-09-21:
Hi does snayone know about an openSCENARIO Editor I can use to create scenarios where in I can easily define routes for the ego vehicle as well. Thanks in advance

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.11]
 - Python version [e.g. 2.7]
 - Version [e.g. 0.9.11]

fabianoboril, 2021-09-21:
https://github.com/carla-simulator/traffic-generation-editor

werewolfdev, 2021-09-21:
Thanks
Will give it a look

xfqbuaa, 2022-05-08:
@fabianoboril 
I download traffic-generation-editor while tried a lot and can not play scenario successfuly. 

The problem is pygame window appear but blackscreen. 
[issue](https://github.com/carla-simulator/traffic-generation-editor/issues/27)

I have checked plugin added, path and also connect to Carla. 
Any suggestions are appreicated. 

[os]: ubuntu 20.04 
[Carla]: 0.9.13

hsiang-lee, 2025-12-04:
You might want to try the newly introduced OpenSCENARIO DSL (OpenSCENARIO 2) standard. Scenario Runner currently offers some support for it, and there's an online tool available for writing scenarios that provides syntax highlighting, auto-completion, and real-time error checking. [osc2studio](https://osc2studio.online)