carla-simulator/scenario_runner #803: Missing support for FollowTrajectoryAction

Briandez, 2021-09-21:
https://github.com/carla-simulator/scenario_runner/blob/05a811a6248c8033e4c76185ff40ae5eccad7e69/srunner/tools/openscenario_parser.py#L1211

Any plans for supporting FollowTrajectoryAction?

glopezdiest, 2021-09-23:
Hey @Briandez, we have PR #751 with the very basic implementation, and it'll be merged soon. The rest of the action has no specific plans, but it'll come eventually

Briandez, 2021-09-23:
Hi @glopezdiest 

I also did something couple of days ago similar to the PR

[image]

      
But somehow i am getting this error on running the scenario:

[image]

But i guess this has something to do with the opendrive itself rather then the implementation but im not sure though.

glopezdiest, 2021-09-23:
How are you passing that trajectory to the actor? Which behavior are you using? Some OSC behaviors modify the actor control, so you need to first create a controller before using those, or the error you are seeing might happen

glopezdiest, 2021-11-23:
Closing this issue while waiting for the original poster to provide more information. Feel free to reopen it if needed

closed 2021-11-23 by glopezdiest