carla-simulator/scenario_runner #804: 'TimeOut' object has no attribute 'optional'

clairebb1005, 2021-09-23:
Hello,

I'm currently running the scenario_runner with openscenario files created by the traffic generation editor. 
But the same error keeps happening..

Scenario 1:  two vehicles(one is an ego vehicle) with initial speeds > 0. Both using Waypoint Maneuvers but without Start Triggers.  
The error shows up when one vehicle reaches the first waypoint.

Scenario 2: only one non-ego vehicle in the scenario with initial speeds > 0, using Waypoint Maneuvers and without Start Triggers. The error shows up before the vehicle shows up. (I guess it's because I need to add an ego vehicle? but is it possible to have my own ego vehicle in a separate file controlled by a reinforcement learning agent?)

Error : 
> Traceback (most recent call last):
>   File "scenario_runner.py", line 408, in _load_and_run_scenario
>     self._analyze_scenario(config)
>   File "scenario_runner.py", line 259, in _analyze_scenario
>     if not self.manager.analyze_scenario(self._args.output, filename, junit_filename, json_filename):
>   File "C:\Users\Desktop\scenario_runner-0.9.11\scenario_runner-0.9.11\srunner\scenariomanager\scenario_manager.py", line 216, in analyze_scenario
>     if (not criterion.optional and
> AttributeError: 'TimeOut' object has no attribute 'optional'
> 'TimeOut' object has no attribute 'optional'
> No more scenarios .... Exiting

Any idea how to solve this problem? Thanks:)

 - OS: Windows 10
 - CARLA Version [0.9.11]
 - Scenario_Runner Version [0.9.11]
 - Python version [3.7]

fabianoboril, 2021-09-23:
Hi,
can you switch to 0.9.11 branch (not tag) of scenario runner? This should resolve this issue.

clairebb1005, 2021-09-23:
ok thanks! solved

closed 2021-09-24 by fabianoboril