carla-simulator/scenario_runner #809: OpenScenario RelativeLanePosition and Waypoint Following

repa1030, 2021-09-24:
**Describe our question or idea**
I have some trouble with OpenScenario with teleporting an adversary vehicle. The teleport works just fine and the vehicle will
spawn on the correct lane but the waypoint follower will not follow this lane but the oncoming lane doing a U-turn ("actual drive path" in picture).

**Expected behavior**
Driving on the current lane where the vehicle was spawned, as shown in the picture below as "expected drive path".

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - CARLA Version 0.9.11
 - Python version 2.7
 - Version 0.9.11

fabianoboril, 2021-09-24:
Can you share the scenario?

repa1030, 2021-09-24:
I just removed the initial position (that was a LanePosition) and now the adversary vehicle starts with the correct drive path but after a while it does a U-turn again.
Here is the scenario file (changed it to txt cause github did not let me upload a xosc file):
[A1_OvertakeOncoming.txt](https://github.com/carla-simulator/scenario_runner/files/7225664/A1_OvertakeOncoming.txt)

fabianoboril, 2021-09-24:
Can you update to the 0.9.11 branch (not tag)? We made a couple of improvements over the time, which might have addressed this issue. The XOSC file looks fine on first glance.

repa1030, 2021-09-24:
I just updated my local repo to the 0.9.11 branch about 1-2 hrs ago 😄 
I corrected the orientation of the RelativeLanePosition, as it was relative before but the problem stays the same:
[A1_OvertakeOncoming.txt](https://github.com/carla-simulator/scenario_runner/files/7225763/A1_OvertakeOncoming.txt)

The vehicle starts driving correct (when no initial pos provided and only the teleport action during the maneuver is given) and 
and after some short time, it does an U-Turn.

WindTi, 2021-12-07:
Did you solve this problem? I had a similar problem when I was writing a scene where the initially generated vehicle was teleport to the new location instead of going along the road but going toward the initial point, as if a new route had been formed between the two points.

WindTi, 2021-12-07:
I'm now on the 0.9.12 branch. I think you have the same problem as I have, I have tried to adjust the adversaries in different positions multiple times, and each time the regenerated vehicle races toward the corresponding initialization build point. Apparently your initialization build point is behind, so you're creating a u-turn.

yiyizi112, 2022-10-23:
> I'm now on the 0.9.12 branch. I think you have the same problem as I have, I have tried to adjust the adversaries in different positions multiple times, and each time the regenerated vehicle races [...]
I have encountered the same problem. Have you solved it?