carla-simulator/scenario_runner #814: Data logging with metrics manager

werewolfdev, 2021-09-28:
Iam getting this error while running python metrics_manager.py --metric srunner/metrics/examples/distance_between_vehicles.py --log srunner/metrics/data/DistanceBetweenVehicles.log
I have added /home/schemba/carla/scenario_runner-0.9.11/srunner/metrics/tools to the python path

Traceback (most recent call last):
File "metrics_manager.py", line 28, in
from srunner.metrics.tools.metrics_log import MetricsLog
ImportError: No module named metrics.tools.metrics_log
Can you kindly provide your insights on the same
**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.11]
 - Python version [e.g. 2.7]
 - Version [e.g. 0.9.11]

werewolfdev, 2021-09-29:
Hi @glopezdiest Any inputs on this issue will be very helpful. Thanks in advance

glopezdiest, 2021-09-29:
I don't seem to be having any errors, and the metrics log is imported correctly. My PYTHONPATH is just the CARLA folder as well as the scenario_runner one

werewolfdev, 2021-09-29:
Thanks . Hmm not sure what Iam doing wrong here then

werewolfdev, 2021-09-29:
I have run a scenario using openSCENARIO, saved the log and json in srunner folder , then kept the world window running and ran the metrics manager script in the scenario runner folder .
Simulation should be running while we call the metrics manager script: This just means we have to have the world up and running using CarlaUE4.sh script right

werewolfdev, 2021-09-30:
Hi Team , if the metrics manager doesnt work is there an alternative to log the data of ego and all target vehicles involved. Im using simple vehicle control so as to control ego vehicle as well.

werewolfdev, 2021-10-04:
The issue was the absence of init python file in the folder since I was using python 2.7.Thanks for the help @glopezdiest

closed 2021-10-04 by werewolfdev