carla-simulator/scenario_runner #819: When will OpenScenario version 1.1.0 be supported?

ChoByungSoo2, 2021-10-06:
hello!
The OpenScenario version has been upgraded. 
When is the open scenario 1.1.0 supported?

fabianoboril, 2021-10-06:
Hi,
I briefly checked the Changelog, and many things should work directly with SR. Is there a particular feature that you are missing?
https://www.asam.net/static_downloads/modelDocumentation/ChangeLog.html

ChoByungSoo2, 2021-10-07:
I use SR version 0.9.11.
https://www.asam.net/static_downloads/modelDocumentation/content/Precipitation.html
https://www.asam.net/static_downloads/modelDocumentation/content/DistanceCondition.html
Like the two above, several have changed, making it difficult to compatible.

Thank you!

fabianoboril, 2021-11-08:
The problem is that 1.1 and 1.0 are not really compatible, or we have to basically support two front-ends. If you are willing to implement the changes, I can support you in guiding what needs to be changed.

bapukokare, 2023-01-31:
@fabuanoboril  pls guide us to support opnescenario 1.1 std with carla