carla-simulator/scenario_runner #826: ModuleNotFoundError: No module named 'srunner'

chinmay-d, 2021-10-15:
I have added SCENARIO_RUNNER_ROOT as senario_runner in the system variables path which I downloaded from [Carla git](https://github.com/carla-simulator/scenario_runner/archive/refs/tags/v0.9.12.zip). In this senario_runner we have srunner folder which I am trying to import to scenario_manager.py and in leaderboard_evaluator.py.  I also added ./senario_runner in settings.json of vscode which helped in removing the warning while importing srunner but still when I try to run leaderboard_evaluator.py it gives me this error 

```
Traceback (most recent call last):
  File "C:\CARLA\leaderboard\leaderboard\leaderboard_evaluator.py", line 27, in <module>
    from srunner.scenariomanager.carla_data_provider import *
ModuleNotFoundError: No module named 'srunner'
```

This is leaderboards_evaluator.py

[image]

This is settings.json

[image]

 - OS: [Windows 10]
 - CARLA Version [CARLA 0.9.12]
 - Python version [e.g. 3.8.5]

I also tried carla on ubuntu but I set MESA intel error and it doesn't run. On windows though it runs but I am getting these import errors which I am trying to solve from some days... 
Can anyone help me with this... I am stuck on setting carla for a long time. Please help if you know where I am going wrong.. 
Thank you :)

glopezdiest, 2021-10-18:
Hey @chinmay-d, it seems like python isn't finding *srunner*, have you added your scenario runner folder to the PYTHONPATH?

chinmay-d, 2021-11-02:
sorry for being late... Yeah I have tried adding scenario_runner and still I get same error...
Also I dualbooted to ubuntu now and tried following instructions on [Leaderboard](https://leaderboard.carla.org/get_started/) I have set leaderboard git and also scenario_runner git for carla_0.9.12 in ubuntu and I still get the same error 

The error:
[image]

VSCode does not show import error:
[image]

My ~/.bashrc:
[image]

And finally the ./test_run.sh:
[image]

I think all my exports are correct and I am getting same error in windows as well as on ubuntu.... If you can find out the error  it will be a lot of help...
Thanks :)

glopezdiest, 2021-11-02:
I see. All your images seem fine and unless your `scenario-runner_0.9.12` doesn't contain `srunner`, I can't see the problem. Could you maybe remove the 3 `srunner` imports and see if the `leaderboard` ones are imported? With that, we can see if the problem is in the PYTHONPATH, or in the srunner module.

chinmay-d, 2021-11-02:
I have commented all the imports containing of srunner and I get this error:
[image]

Yeah I think its a PYTHONPATH issue, but can't figure out what's wrong..

chinmay-d, 2021-11-02:
I guess, the python version in `PythonAPI/carla/dist` is 3.7 but the 'carla' env that I made using pyenv has version 3.8.0. As carla 0.9.12 supports python version 3.8 so I used an env with python version 3.8.0. Is this causing the problem?

chinmay-d, 2021-11-02:
@glopezdiest  I even tried with env python version 3.7.1 I still get same above error `ModuleNotFoundError: No module named 'agents.navigation.global_route_planner_dao'` after commenting all the srunner imports

glopezdiest, 2021-11-02:
That might be another issue, as we changed the agents for 0.9.12, and without [these](https://github.com/carla-simulator/leaderboard/commit/56484354a019a3094b5c0bbf62beb0642c2fa559) changes, the leaderboard crashes

chinmay-d, 2021-11-02:
@glopezdiest, so I should try cloning leaderboard again?

chinmay-d, 2021-11-02:
yea, I made the changes and I still get `ModuleNotFoundError: No module named 'srunner'` and when I comment all the imports of srunner from all .py files I get this error:
[image]

chinmay-d, 2021-11-02:
I closed my terminal and opened a new terminal and uncommented all the imports of srunner.... and It worked....
All I did after cloning git of [leaderboard](https://github.com/carla-simulator/leaderboard) when I got support for carla 0.9.12 is cloned git of [scenario_runner-0.9.12](https://github.com/carla-simulator/scenario_runner/releases/tag/v0.9.12) and restarted the teriminal.... and now its working..
Thanks @glopezdiest  :)

GeneralMak, 2021-11-03:
@chinmay-d  Hello ,can you tell me how to successfuly run this leaderboard_evaluator.py?  thank you

glopezdiest, 2021-11-03:
Oh, seems like you updated the PYTHONPATH but didn't refresh the terminal nor sourced the new bash script, so the new PYTHONPATH wasn't being used. It should work for you now.

@GeneralMak Please, open another issue with your specific issues.

closed 2021-11-03 by glopezdiest