carla-simulator/scenario_runner #828: an error occurred when using the metrics module

GeneralMak, 2021-11-02:
when I use the following commands  to run scenario ,there will always be an error.
_1.python scenario_runner.py --scenario FollowLeadingVehicle_1 --record srunner/metrics/data_
__2.python3 metrics_manager.py --metric srunner/metrics/distance_between_two_vehicles.py --log srunner/metrics/data/FollowLeadingVehicle_1.log_
[image]

Save the child,thank you very much!
**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - CARLA Version [e.g. CARLA 0.9.12]
 - Python version [e.g. 3.7]
 - Version [e.g. 0.9.12]

GeneralMak, 2021-11-02:
I refer to what the official document says ,it can't run successfully

GeneralMak, 2021-11-03:
> when I use the following commands to run scenario ,there will always be an error. _1.python scenario_runner.py --scenario FollowLeadingVehicle_1 --record srunner/metrics/data_ __2.python3 [...]

GeneralMak, 2021-11-03:
es.githubusercontent.com/56059626/139816294-69a3c17c-ca7f-490d-b0a0-17beb7bbfce4.jpg)
> 
> Save the child,thank you very much! **Desktop (please complete the following information):**
> 
> * OS: [e.g. Windows]
> * CARLA Version [e.g. CARLA 0.9.12]
> * Python version [e.g. 3.7]
> * Version [e.g. 0.9.12]

GeneralMak, 2021-11-03:
I successfully recorded a FollowLeadingVehilce_1 by running _python scenario_runner.py --scenario FollowLeadingVehicle_1 --record srunner/metrics/data_,but when I run the second command(_python3 metrics_manager.py --metric srunner/metrics/distance_between_two_vehicles.py --log srunner/metrics/data/FollowLeadingVehicle_1.log_), I encountered the following error, do you know how to solve it?  
[image]

I imported __init()__.py file.

GeneralMak, 2021-11-08:
When I use the Disance_between_vehilces metric for FollowLeadingVehilce_1.log, there will be AttributeError"'NoneType' object has no attribute'location',
When I use the distance_to_lane_center metric for ControlLoss_1.log, I still get AttributeError"'NoneType' object has no attribute'location'. Then I print(log.get_actor_transform(ego_id,i)) in the 56th line of the disaster_between_vehilces and the output is None? But I get ego vehicle's id through print(ego_id)
Why is this? 
[image]
@glopezdiest  Thanks in advance!!!

GeneralMak, 2021-11-08:
> When I use the Disance_between_vehilces metric for FollowLeadingVehilce_1.log, there will be AttributeError"'NoneType' object has no attribute'location', When I use the distance_to_lane_center [...]

glopezdiest, 2021-11-09:
I'll take a look, seems like the metrics module is broken

GeneralMak, 2021-11-09:
> I'll take a look, seems like the metrics module is broken

Is the offical metrics module broken? or my metrics module is broken

glopezdiest, 2021-11-09:
Seems like the metrics module itself, but I'm not sure as I haven't used it in a while

GeneralMak, 2021-11-10:
> Seems like the metrics module itself, but I'm not sure as I haven't used it in a while

Do you know how to solve this problem?

glopezdiest, 2021-11-10:
@GeneralMak I'm linking this issue with #829, as you already answered there and the error is basically the same. We'll close this when the other one is resolved too.

closed 2024-06-14 by glopezdiest