carla-simulator/scenario_runner #829: Logging Velocities and Acceleration with scenario_runner.py

joluko, 2021-11-03:
Hello,
I want to plot the Velocitie and Accelerations from the Ego Vehicle form the log file createtd with the scenario_runner.py. 
Therefore I am using the metrics_manager. The Problem is that i.e. log.get_actor_acceleration(self, actor_id, frame) always returns 'None'.

I think i have to edit the recorder somehow, because the velocties and accelerations are not logged but i dont know how. 
Can someone please help?

 - OS: Windows 10 64 bit
 - CARLA Version 0.9.11
 - Python version 3.7.11
 - Version 0.9.11

GeneralMak, 2021-11-08:
Yes，I currently hava a problem  similar to yours ,When using the log.get_actor_transform(self, actor_id, frame) always return 'None'

glopezdiest, 2021-11-09:
The idea would be that if the actor_id is correct, then you'd get the transform of the actor, but it might be broken, as pointed out by another issue. I'll test it an answer asap

glopezdiest, 2021-11-10:
@GeneralMak @joluko After some testing, I can't really see your error. I tested the metrics module **for version 0.9.11 everything is working fine**. Running the distance between vehicles examples:
```
python3 metrics_manager.py --metric srunner/metrics/examples/distance_between_vehicles.py --log srunner/metrics/data/DistanceBetweenVehicles.log
```
I can definitely see the graph created with matplotlib, and `log.get_actor_transform(self, actor_id, frame)` does not return None for me.

The metrics module is broken for 0.9.12, failing to parse some wheel's data plus the examples log don't seem to be working. However, those two are minor errors and once fixed, everything seems to be working just fine, and the metric again, show the desired graph.

Could you both explain exactly what you are running, and if needed, attach the files?

GeneralMak, 2021-11-11:
> @GeneralMak @joluko After some testing, I can't really see your error. I tested the metrics module and **for version 0.9.11 everything is working fine**. Running the distance between vehicles examples:
> 
> ```
> python3 metrics_manager.py --metric srunner/metrics/examples/distance_between_vehicles.py --log srunner/metrics/data/DistanceBetweenVehicles.log
> ```
> 
> I can definitely see the graph created with matplotlib, and `log.get_actor_transform(self, actor_id, frame)` does not return None for me.
> 
> The metrics module is broken for 0.9.12, failing to parse some wheel's data plus the examples log don't seem to be working. However, those two are minor errors and once fixed, everything seems to be working just fine, and the metric again, show the desired graph.
> 
> Could you both explain exactly what you are running, and if needed, attach the files?

Okay, first of all thank you very much for your help.

I do is to create some scenarios by myself, and get some data through the metric module, such as distance between cars, speed, acceleration, etc.
Will you re-release the revised version of the measurement module in 0.9.12? When will it be released?

GeneralMak, 2021-11-11:
I also tried it in Carla 0.9.10 and  it worked correctly. So I decided to use Carla 0.9.10 for the time being

glopezdiest, 2021-11-11:
> Will you re-release the revised version of the metric module in 0.9.12? When will it be released?

I'll update it today, and it'll be ready for 0.9.13. Anyway, the changes are very minor, so you should be able to cherry pick it to 0.9.12.

GeneralMak, 2021-11-11:
> > Will you re-release the revised version of the metric module in 0.9.12? When will it be released?
> 
> I'll update it today, and it'll be ready for 0.9.13. Anyway, the changes are very minor, so you should be able to cherry pick it to 0.9.12.

Okay, thank you very much for revising the carla 0.9.12 metric module! In addition, I have a question to ask you, why can get the speed of ego vehicles through _**log.get_actor_velocity(actor_id, frame)**_ from DistanceBetweenVehilces.log. But when I get the log file from the scenario I created, the return value through **_log.get_actor_velocity(actor_id, frame)_** is None? I can’t get the speed, why is it so?

GeneralMak, 2021-11-15:
Hello, I found that after updating the metic_module you corrected, the original problem not only has not been solved, but the problem has become more serious. Before, I could use ScenarioRunner-0.9.12 to record scenarios, but I could not measure scenarios, but now I can neither record scenarios nor measure scenarios
The following error is always reported when recording scenes:
[image]
Always fail to connect to Carla server when measuring scenes
[image]
@glopezdiest  Thanks in advance!

glopezdiest, 2021-11-16:
@GeneralMak Sorry for the delay, I'll take a look. Maybe we changed the recorder, and it breaks if you mix different versions? All my tests where using new logs, so I'll do more digging and report back.

fly12312312, 2021-11-16:
Hello, I saw your changes to metrics_module in ScenarioRunner 0.9.12, and I did so. But I also encountered the  problem. When running _python metrics_manager.py --metric srunner/metrics/examples/distance_between_vehicles.py --log srunner/metrics/data/DistanceBetweenVehicles.log_, it appears that'NoneType' object has no attribute'location' As shown in the figure below 。
[image]

    OS: Windows 10 64 bit
    CARLA Version 0.9.12
    Python version 3.7
    Version 0.9.12

@glopezdiest Thanks in advance
@GeneralMak Have you solved your problem?

GeneralMak, 2021-11-16:
@glopezdiest  
First of all, thank you very much for your reply. I have solved the problem of scenario recording(). I found that it is necessary to set _**True**_ in line 408 of **_scenario_runner.py_** to **_False_** to record the scene. As shown below.
[image]
It successfully recorded the scenario!
[image]
Secondly, for the scene parameters recorded by the metric, I found that if the line 77 in **_metric_anager.py_** is **_True_**, the connection to the Cara server will time out. If it is set to **_False_**, the error shown in the figure below will be reported, and @ fly12312312 shows the same.
[image]

GeneralMak, 2021-11-16:
At present, I still don’t know why changing the values of True and False will result in different changes.

glopezdiest, 2021-11-16:
@GeneralMak @fly12312312 I'm just oh so stupid. For 0.9.12, we added the large maps feature, which changed the recorder, so the example _logs_ don't work. This was causing the `client.show_recorder_file_info()` to fail, having missing data or timing out, which explains the issues you are seeing. In other words, **you can't use the examples *logs* with CARLA 0.9.12, but you can use newly generated ones**.

Can you confirm that this is consistent with what you are seeing?

@GeneralMak About your doubts:
> I found that it is necessary to set True in line 408 of scenario_runner.py to False to record the scene

This was set to True when the metrics module was introduced, as it makes the recorder save all the simulation data. Without it, only the information needed to replay the scene is saved, which we don't want.

> Secondly, for the scene parameters recorded by the metric, I found that if the line 77 in metric_anager.py is True, the connection to the Cara server will time out

I also saw this, but this doesn't happen for all logs, some just are incorrectly parsed. It's probably a bug when trying to use different versions, and new logs work fine. Again, this should be set to True to get all the simulation information.

PD: I'll generate new example logs if we confirm this is the issue

GeneralMak, 2021-11-17:
@glopezdiest Thank you very much for your answers. I understand, what you mean is that there is no need to modify the _**True**_ value in scenario_runner.py and metric_manager.py. The reason why the scene cannot be recorded and measured is because of the Carla 0.9.12 version.

PR #842 (2021-11-25): Compatibility with CARLA 0.9.13
### Description

- Changed the version of SR to 0.9.13. SR still checks for CARLA's version 0.9.12 or higher though, as SR 0.9.13 is compatible with 0.9.12
- Remade the metrics examples, and added some checks for some special actors

Fixes #829 and #828

#### Where has this been tested?

  * **Platform(s):** Ubutnu 18.04
  * **Python version(s):** 3.6
  * **Unreal Engine version(s):** CARLA's "4.26"
  * **CARLA version:** 0.9.13

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/842)
closed 2021-11-25 by fabianoboril