carla-simulator/scenario_runner #831: list index out of range when running Traffic Scenario 04

youngzhou1999, 2021-11-08:
Hi, there.
I got an error when running routes 6 in Town01 of traning.xml and scenario04 of all_scenarios.json. The whole message looks like this(ignored file path):
`Traceback (most recent call last):
  File "srunner/scenarios/route_scenario.py", line 401, in _build_scenario_instances
    world, [ego_vehicle], scenario_configuration, criteria_enable=False, timeout=timeout
  File "srunner/scenarios/object_crash_intersection.py", line 479, in __init__
    criteria_enable=criteria_enable)
  File "srunner/scenarios/basic_scenario.py", line 52, in __init__
    self._initialize_actors(config)
  File "srunner/scenarios/object_crash_intersection.py", line 487, in _initialize_actors
    waypoint = generate_target_waypoint_in_route(self._reference_waypoint, self._ego_route)
  File "/srunner/tools/scenario_helper.py", line 369, in generate_target_waypoint_in_route
    route_location = route[index][0]
IndexError: list index out of range
`
It seems that the loop in line 397 of `scenario_helper.py` got overflow? And why didn't the loop got break?

glopezdiest, 2021-11-10:
Hey @youngzhou1999, could you explain with a bit more detail what you are doing? I just run 
```
python3 scenario_runner.py --route srunner/data/routes_training.xml srunner/data/all_towns_traffic_scenarios.json 6 --agent srunner/autoagents/human_agent.py
```
and everything seems to run correctly

youngzhou1999, 2021-11-11:
Hey @glopezdiest , thanks for reply.
I was testing the behavior of my own policy running on Town01. So the error was caused by the function called `generate_target_waypoint_in_route` in Scenario04 `VehicleTurningRoute`. I test the `auto_agent` just now and it seems fine. So maybe there are some problems in my policy.

closed 2021-11-15 by youngzhou1999