carla-simulator/scenario_runner #832: OpenScenario parameter assignment is broken

lasuomela, 2021-11-08:
**Describe the bug**
A clear and concise description of what the bug is.

Hello, I updated the scenario runner from 
[05a811a](https://github.com/carla-simulator/scenario_runner/commit/05a811a6248c8033e4c76185ff40ae5eccad7e69)
to
[e5b06cc](https://github.com/carla-simulator/scenario_runner/commit/e5b06cc727831ffe61ff828e0566c202ef34ae04)

to get OpenScenario ParameterAction support. However, now it seems that the parsing of .xosc files is somehow broken. When a parameter is used in the .xosc file, the name of a parameter is passed instead of the value assigned to a parameter.

**To Reproduce**
Steps to reproduce the behavior:
1. Declare a parameter in .xosc scenario file:
```
<?xml version="1.0"?>
<OpenSCENARIO>
  <FileHeader revMajor="1" revMinor="0" date="2021-05-04T00:00:00" description="CARLA:Template" author=""/>
  <ParameterDeclarations>
    <ParameterDeclaration name="route_entry_name" parameterType="string" value="Map01_testroute_maneuver"/>
  </ParameterDeclarations>
```

2. Use the parameter for something, e.g. retrieve a maneuver from catalog:
```
<ManeuverGroup maximumExecutionCount="1" name="ManeuverSequence">
  <Actors selectTriggeringEntities="false">
    <EntityRef entityRef="ego_vehicle"/>
  </Actors>
  <CatalogReference catalogName="ManeuverCatalog" entryName="$route_entry_name"/>
</ManeuverGroup>
```

**Expected behavior**
Before updating, this retrieved the catalog maneuver with name "Map01_testroute_maneuver" correctly.

**Screenshots**
Leads to:
[image]
This seems to affect all parameter usage in the .xosc file. Before updating the scenario runner everything was working well. I suspect the breaking changes are introduced by [8ed24c8](https://github.com/carla-simulator/scenario_runner/commit/8ed24c8da79d27dfbfd7fe00c12802eaf25f8b20), however I couldn't find the cause.

**Desktop (please complete the following information):**
 - OS: [Ubuntu 20.04
 - CARLA Version 0.9.12 (docker shipped)
 - Python version 3.7

I'm using the scenario runner through the ros-bridge.

**Additional context**
Add any other context about the problem here.

lasuomela, 2021-11-08:
Thanks for the quick reply. I tried the new fix, with that the ouput is 
[image]
So the problem still persists

fabianoboril, 2021-11-08:
Can you share your scenario file? The traceback is different now...

lasuomela, 2021-11-08:
So I managed to solve the error message for the route maneuver catalog, resolving the error message from my previous comment by adding a string conversion to the line from your fix:

```
         entry = catalogs[catalog_reference.attrib.get("catalogName")][ParameterRef(catalog_reference.attrib.get("entryName"))]
```
->
```
         entry = catalogs[catalog_reference.attrib.get("catalogName")][str(ParameterRef(catalog_reference.attrib.get("entryName")))]
```

However, after solving I'm running into similar issues with the rest of the parameters too:

[image]

 This too was solved by adding ParameterRef in [openscenario_parser](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/tools/openscenario_parser.py) get_weather_from_env_action(), line 477:

```
        carla_weather.sun_altitude_angle = math.degrees(float(sun.attrib.get('elevation', 0)))
```
->
```
        carla_weather.sun_altitude_angle = math.degrees(float(ParameterRef(sun.attrib.get('elevation', 0))))
```
But the openscenario_parser.py seems to be full of this kind of catalog refs, meaning adding this in a lot places. Should this be addressed here, or on a higher level? As I said, my scenario file was working fine with the earlier commit of scenario runner, so I think there was a change introduced somewhere that causes the parameter parsing to not properly work in the master. 

Also, the problem doesn't just affect catalogs.

This works:
```
<?xml version="1.0"?>
<OpenSCENARIO>
  <FileHeader revMajor="1" revMinor="0" date="2021-05-04T00:00:00" description="CARLA:VisualNavigatorTemplate" author=""/>
  <ParameterDeclarations>
    <ParameterDeclaration name="town_name" parameterType="string" value="Town01"/>
  </ParameterDeclarations>
  <CatalogLocations>
  </CatalogLocations>
  <RoadNetwork>
    <LogicFile filepath="Town01"/>
    <SceneGraphFile filepath=""/>
  </RoadNetwork>
```
This doesn't:
```
<?xml version="1.0"?>
<OpenSCENARIO>
  <FileHeader revMajor="1" revMinor="0" date="2021-05-04T00:00:00" description="CARLA:VisualNavigatorTemplate" author=""/>
  <ParameterDeclarations>
    <ParameterDeclaration name="town_name" parameterType="string" value="Town01"/>
  </ParameterDeclarations>
  <CatalogLocations>
  </CatalogLocations>
  <RoadNetwork>
    <LogicFile filepath="$town_name"/>
    <SceneGraphFile filepath=""/>
  </RoadNetwork>
```
So I think it's a general bug related to parameter parsing.

Here is my scenario file for reference anyway. I will be deleting it once we resolve this issue.

```
<?xml version="1.0"?>
<OpenSCENARIO>
  <FileHeader revMajor="1" revMinor="0" date="2021-05-04T00:00:00" description="CARLA:Template" author=""/>
  <ParameterDeclarations>
    <ParameterDeclaration name="town_name" parameterType="string" value="Town01"/>
    <ParameterDeclaration name="global_extractor_name" parameterType="string" value="XXX"/>
    <ParameterDeclaration name="local_extractor_name" parameterType="string" value="XXX"/>
    <ParameterDeclaration name="local_matcher_name" parameterType="string" value="XXX"/>
[... 256 lines cut ...]
          </ConditionGroup>
          <ConditionGroup>
            <Condition name="logger_act_success_stopcondition" delay="0" conditionEdge="rising">
              <ByValueCondition>
                <StoryboardElementStateCondition storyboardElementType="event" storyboardElementRef="success_log_event" state="completeState"/>
              </ByValueCondition>
            </Condition>
          </ConditionGroup>
        </StopTrigger>
        
      </Act>
    </Story>
    <StopTrigger>
      <ConditionGroup>
        <Condition name="scenario_endcondition" delay="0" conditionEdge="rising">
          <ByValueCondition>
            <StoryboardElementStateCondition storyboardElementType="act" storyboardElementRef="Loggers" state="completeState"/>
          </ByValueCondition>
        </Condition>
      </ConditionGroup>
    </StopTrigger>
  </Storyboard>
</OpenSCENARIO>

```

lasuomela, 2021-11-08:
Actually, here's the minimum steps to reproduce, based on making the map choice a parameter in the FollowLeadingVehicle.xosc example (changes only on lines 6 and 11, compared to the original):

```
<?xml version="1.0"?>
<OpenSCENARIO>
  <FileHeader revMajor="1" revMinor="0" date="2019-06-25T00:00:00" description="CARLA:FollowLeadingVehicle" author=""/>
  <ParameterDeclarations>
    <ParameterDeclaration name="leadingSpeed" parameterType="double" value="2.0"/>
    <ParameterDeclaration name="townName" parameterType="string" value="Town01"/>
  </ParameterDeclarations>
  <CatalogLocations>
  </CatalogLocations>
  <RoadNetwork>
    <LogicFile filepath="$townName"/>
    <SceneGraphFile filepath=""/>
  </RoadNetwork>
  <Entities>
    <ScenarioObject name="ego_vehicle">
      <Vehicle name="vehicle.lincoln.mkz2017" vehicleCategory="car">
        <ParameterDeclarations/>
        <Performance maxSpeed="69.444" maxAcceleration="10.0" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties>
          <Property name="type" value="ego_vehicle"/>
          <Property name="color" value="0,0,255"/>
        </Properties>
      </Vehicle>
    </ScenarioObject>
    <ScenarioObject name="adversary">
      <Vehicle name="vehicle.lincoln.mkz2017" vehicleCategory="car">
        <ParameterDeclarations/>
        <Performance maxSpeed="69.444" maxAcceleration="10.0" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties>
          <Property name="type" value="simulation"/>
          <Property name="color" value="255,0,0"/>
        </Properties>
      </Vehicle>
    </ScenarioObject>
  </Entities>
  <Storyboard>
    <Init>
      <Actions>
        <GlobalAction>
          <EnvironmentAction>
            <Environment name="Environment1">
              <TimeOfDay animation="false" dateTime="2019-06-25T12:00:00"/>
              <Weather cloudState="free">
                <Sun intensity="1.0" azimuth="0.0" elevation="1.31"/>
                <Fog visualRange="100000.0"/>
                <Precipitation precipitationType="dry" intensity="0.0"/>
              </Weather>
              <RoadCondition frictionScaleFactor="1.0"/>
            </Environment>
          </EnvironmentAction>
        </GlobalAction>
[... 134 lines cut ...]
        </Condition>
        <Condition name="criteria_KeepLaneTest" delay="0" conditionEdge="rising">
          <ByValueCondition>
            <ParameterCondition parameterRef="" value="" rule="lessThan"/>
          </ByValueCondition>
        </Condition>
        <Condition name="criteria_CollisionTest" delay="0" conditionEdge="rising">
          <ByValueCondition>
            <ParameterCondition parameterRef="" value="" rule="lessThan"/>
          </ByValueCondition>
        </Condition>
        <Condition name="criteria_DrivenDistanceTest" delay="0" conditionEdge="rising">
          <ByValueCondition>
            <ParameterCondition parameterRef="distance_success" value="100" rule="lessThan"/>
          </ByValueCondition>
        </Condition>
      </ConditionGroup>
    </StopTrigger>
  </Storyboard>
</OpenSCENARIO>
```

Running this through the ros-bridge rviz2-plugin leads to
[image]
on my machine.

fabianoboril, 2021-11-09:
Hi,
I pushed a new PR that addresses this problem. In fact, there were some considerable changes to enable OSC ParameterActions, that allow changing parameters at runtime. Unfortunately, that required quite some rework under the surface of scenariorunner. I hope I fixed all relevant locations. I tested it with your sample code, which seems to work now.

lasuomela, 2021-11-09:
Hi,
I was working on the same changes. I cross-checked our changes to make sure most cases are handled. I also updated the ParameterRef class to handle other data types than float.

But one more thing. With the changes you suggested, the ParameterAction wasn't working. This was because the config file is parsed at scenario init, so any parameters that would be affected by ParameterAction were actually hard coded at init. Because of this, ParameterAction correctly modified the parameter values stored by  CarlaDataProvider, but these new values weren't accessed when evaluating Actions such as UserDefinedAction.

I resolved this for UserDefinedAction by moving parameter value resolving into RunScript.update() in atomic_behaviors.py, which is evaluated at runtime. Changes made into parameter values by ParameterAction are now correctly reflected in parameters supplied to UserDefinedAction. I don't know if this issue comes up when using parameters affected by ParameterAction with Actions types other than UserDefinedAction. To avoid circular imports, I had to move ParameterRef definition into it's own file. The solution isn't perfect, but this PR should extend the support for ParameterAction atleast a bit.

Please comment on style and naming conventions if necessary, I haven't contributed before. Thanks for your attention on the issue, this really helped me a lot!

fabianoboril, 2021-11-09:
Hi,
ah good to see that you went into the code as well. What I am not getting from your statement: Are you saying my new PR is breaking ParameterActions? It shouldn't, but I haven't tested that part yet. It was just a quick fix to get your things working.

In the past, we went over the XOSC file and basically replaced all parameter calls with the default values. However, to enable dynamic parameters, the ParameterRef class was introduced. The problem that arises is that all xml-tree accesses should be routed through ParameterRef (aka attrib.get(value) --> ParameterRef(attrib.get(value))). This was forgotten for some places, and my PR intended to fix this.

lasuomela, 2021-11-09:
Hi,
Looking at ParameterRef implementation, it seems to me that running str(ParameterRef(parameter)) retrieves the value of the parameter held in CarlaDataProvider at the time of the call. Since this is now being called in openscenario_parser that is run once at the initialization of scenario, it means that the xmltree values are hard coded to the parameter values present at the initialization of the scenario, and ParameterAction has no effect on the xml tree.

This is why I moved the ParameterRef for UserDefinedAction into the atomic behaviour update() method. That is evaluated at runtime, not at scenario initialization. I tried the both ways, and this was the one that worked. Calling ParameterRef in openscenario_parser led to the parameter value in the xmltree staying the same despite ParameterAction.

I suppose other maneuvers also might need this modification?

fabianoboril, 2021-11-09:
Well, the problem is that we construct the entire behavior tree for the scenario at the very beginning. If that construction fails, because a ParameterRef has to be converted to float/str/whatever, we have to do the conversion. But then, even if this is very late in the scenario, the value is already converted and may not be updated. That basically would mean that everything that can be a parameter has to be fowarded to the corresponding atomics and interpreted at runtime.

lasuomela, 2021-11-09:
Yeah, I agree with you that fully supporting ParameterActions would require everything that can be parametrized to be forwarded onto the corresponding atomic behavior.

Is this something that would be desirable?

fabianoboril, 2021-11-09:
I guess, if we touch it now, it is better to do it right ;) Do you want to give it a try?

lasuomela, 2021-11-13:
Hmm. What would it require on code level? From how I understand the code, this would mean doing the same thing I did for UserDefinedAction for 

1. atomic behavior constructions in openscenario_parser/convert_maneuver_to_atomic()
2. atomic condition constructions in openscenario_parser/convert_condition_to_atomic()
3. Reworking openscenario_parser functions such as get_weather_from_env_action() into functions that are evaluated at runtime, in a similar fashion to the the atomi conditions/behaviors
4. Something more?

All of this seems a bit much to do by myself, but I could do one piece for starters. For example rework the functions in convert_maneuver_to_atomic().

lasuomela, 2021-12-23:
Hello @fabianoboril, I began implementing the runtime parameterRef resolution for atomic behaviours. I started by implementing the functionality for AbsoluteTargetSpeed and RelativeTargetSpeed OSC instructions. Could you check the included commit to determine if my approach is okay?

The commit can be found [here](https://github.com/lasuomela/scenario_runner/commit/0813aa65992e834ad41413e91de8022821ecef24?diff=unified).

The relevant changes can be found in openscenario_parser.py lines 1051-1089
and atomic_behaviors.py lines 443-662 (class ChangeActorTargetSpeed(AtomicBehavior)).

Please let me know if this seems good, and I'll implement the changes for the rest of the atomic behaviors.

lasuomela, 2022-01-07:
Hello @fabianoboril ! (Tagging @germanros1987 in case he's not actively working on Carla at the moment.) 

PR #834 now intoduces ParameterAction support to all OpenSCENARIO Actions - the parameter values are now resolved during runtime inside the AtomicBehaviors. I tested each of the actions with ParameterAction and they now seem to work correctly. Could you review the changes, please?