carla-simulator/scenario_runner #839: scenrio_runner crashes server on CARLA 0.9.12

MaddysonBlaze, 2021-11-16:
I have been trying to run a simulation on using OpenScenario files and scenario runner. Please find my local setup details below.

CARLA version: 0.9.12 precompiled version and source build version

Platform/OS: **Ubuntu 18.04 LTS**

GPU : **NVIDEA GeFORCE RTX 3070**

Graphics driver : **Nvidia driver version : 495.44**

Scenario Runner version : **0.9.12_RSS**

Problem you have experienced:

I get the below console output when i run scenario_runner.py script

      ```
scenario_runner main function
      API Version: 0.9.12
      Segmentation fault (core dumped)

```
And the carla server crashes with the console output as shown below
      
```
LowLevelFatalError [File:Unknown] [Line: 803]
VulkanRHI::vkGetPhysicalDeviceSurfaceFormatsKHR(Device.GetPhysicalHandle(), Surface, &NumFormats, nullptr) failed, VkResult=-3
at /home/jenkins/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanSwapChain.cpp:179
with error VK_ERROR_INITIALIZATION_FAILED
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744
Malloc Size=131160 LargeMemoryPoolOffset=327928
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

Steps to reproduce:

run carla server using the command `./CarlaUE4.sh`
run scenario_runner.py using the command `python scenario_runner.py --openscenario /path/to/xosc/file`
I tried launching Carla server with vulkan attribute as well : `./CarlaUE4.sh -vulkan`
But i get the same error with this as well.

Please let me know if any further details are needed.

glopezdiest, 2021-11-23:
Hey @MaddysonBlaze, could you provide the exact commands you are using to encounter this error.? And is this happening immediately or after a while? Debugging server issues with packages is next to impossible so any information you can provide will be appreciated

roque2205, 2021-12-06:
Hi, I get the same error when I try to run FollowLeadingVehicle_1. It works when I set my custom map as scenario town, so it seems to be a map-related issue?

roque2205, 2022-02-08:
For me the scenario runner works perfectly since updating to 0.9.13.

klaus99815, 2022-04-07:
> 对我来说，自从更新到0.9.13以来，场景运行器工作得很好。

Do you try to run Carla iteratively?
Carla crashes when I try to run Carla and Sr iteratively about 30 times....

MahirGulzar, 2023-10-24:
Also, have the same problem. I was successfully running leaderboard-1.0 with and Carla_0.9.10.1 with leaderboard branch of CARLA scenario_runner. I was moved to Carla_0.9.13. I started seeing this crash right after first scenario is evaluated.

xTayEx, 2024-01-06:
> > 对我来说，自从更新到0.9.13以来，场景运行器工作得很好。
> 
> Do you try to run Carla iteratively? Carla crashes when I try to run Carla0.9.13 and SR iteratively about 30 times....

Exactly what I met. I use use scenario_runner to run a scenario iteratively in order to get fitness function value in GA. The carla server will crash with signal 6 after several times run.