carla-simulator/scenario_runner #840: Error when trying to run distance_between_vehicles.py with the metrics module and my own logfile

Brandeyy, 2021-11-17:
I am getting this error message when trying to run distance_between_vehicles.py and I cant figure out where my mistake is

[image]

 - OS: [Windows10]
 - CARLA Version [e.g. CARLA 0.9.11]
 - Python version [e.g. 2.7]
 - Version [e.g. 0.9.11]

glopezdiest, 2021-11-18:
Hey @Brandeyy I've never seen that before, can you pass me the *.log* file? I'll try to run it, see what's happenning

Brandeyy, 2021-11-18:
@glopezdiest this is the log file, this also happened with other log files I recorded myself. The example one worked just right. 
[VehicleConstructionSite.log](https://github.com/carla-simulator/scenario_runner/files/7563993/VehicleConstructionSite.log)

glopezdiest, 2021-11-23:
After some digging, it seems like the spectator camera, as well as sensors and other objects, has a bounding box of 'nan' (and an extent of 'inf' (`Id: 432 origin: (-nan, -nan, -nan) extension: (-inf, -inf, -inf)`)

The weird thing is that does type of string should be correctly parsed and is indeed what's happening to me, as I just run your test and it didn't fail. Anyway, I'll just add a check for that case, so that it doesn't happen, as the bounding box of these actors is pretty much irrelevant. For your case, you should check [this function](https://github.com/carla-simulator/scenario_runner/blob/v0.9.11/srunner/metrics/tools/metrics_parser.py#L187) and do the same, something like:
```py
def parse_bounding_box(info):
    if 'nan' in info[3]:
        location = carla.Location()
    else:
        location = carla.Location(
            float(info[3][1:-1])/100,
            float(info[4][:-1])/100,
            float(info[5][:-1])/100,
        )

    if 'inf' in info[7]:
        extent = carla.Vector3D()
    else:
        extent = carla.Vector3D(
            float(info[7][1:-1])/100,
            float(info[8][:-1])/100,
            float(info[9][:-1])/100,
        )

    return carla.BoundingBox(location, extent)
```

closed 2021-11-23 by glopezdiest

Brandeyy, 2021-11-24:
Thanks a lot !