carla-simulator/scenario_runner #845: Latest CARLA 0.9.13-compatible branch doesn't work with Python2 (ROS1 Melodic)

xqrdot, 2021-11-26:
# Overview
ScenarioRunner has compatibility issues with Python2. Some syntax is Python3 dependant.

# Tracebacks
Here are the Tracebacks of different issues:

## Case 1: srunner/tools/openscenario_parser.py
```
~scenario_runner on master ➜ python2 scenario_runner.py --scenario FollowLeadingVehicleWithObstacle_1 --reloadWorld

Traceback (most recent call last):
  File "scenario_runner.py", line 35, in <module>
    from srunner.scenarioconfigs.openscenario_configuration import OpenScenarioConfiguration
  File "/home/xqr/dev/src/config/scenario_runner/srunner/scenarioconfigs/openscenario_configuration.py", line 24, in <module>
    from srunner.tools.openscenario_parser import OpenScenarioParser, ParameterRef
  File "/home/xqr/dev/src/config/scenario_runner/srunner/tools/openscenario_parser.py", line 96
    def __init__(self, reference_text) -> None:
                                       ^
SyntaxError: invalid syntax
```
Specifying return type is not supported in Python2. All of those can be removed, including this:
```py
# line:113
    def _is_matching(self, pattern: str):
# change to:
    def _is_matching(self, pattern):
```

## Case 2: srunner/scenariomanager/watchdog.py
Relates to #757 

If you fix issues from Case 1, then there is a `watchdog.py` module that is expected to be run with Python3 to greet you:
```
scenario_runner on master ➜ python2 scenario_runner.py --scenario FollowLeadingVehicleWithObstacle_1 --reloadWorld
Traceback (most recent call last):
  File "scenario_runner.py", line 37, in <module>
    from srunner.scenariomanager.scenario_manager import ScenarioManager
  File "/home/xqr/dev/src/config/scenario_runner/srunner/scenariomanager/scenario_manager.py", line 23, in <module>
    from srunner.scenariomanager.watchdog import Watchdog
  File "/home/xqr/dev/src/config/scenario_runner/srunner/scenariomanager/watchdog.py", line 14, in <module>
    import simple_watchdog_timer as swt
  File "/home/xqr/.local/lib/python2.7/site-packages/simple_watchdog_timer.py", line 11
    def __init__(self, callback, check_interval_sec: float = 0.01, trigger_delta_sec: float = 1, identifier=None):
                                                   ^
SyntaxError: invalid syntax
```
Fix is taking this file fully from `0.9.11`: https://github.com/carla-simulator/scenario_runner/blob/v0.9.11/srunner/scenariomanager/watchdog.py

# Resolution

With those simple fixes, I could run `scenario_runner.py` with `python2`. 

Hence, launching a `carla_ros_scenario_runner` is now possible with `roslaunch carla_ros_scenario_runner carla_ros_scenario_runner.launch`

# PC information
 - OS: Ubuntu 18.04 LTS
 - CARLA 0.9.13
 - ROS1 Melodic
 - Python 2.7
 - Version: `master` branch (`85109e7`)

fabianoboril, 2021-11-29:
This is known, and was mentioned in the Changelog. SR is no longer compatible with Python 2 (which is EOL).

xqrdot, 2021-11-29:
> This is known, and was mentioned in the Changelog. SR is no longer compatible with Python 2 (which is EOL).

Ah, didn't see that it is mentioned in the changelog. Makes sense then, have nothing against that decision. This Issue will come in handy for anyone stuck with Python2 too.

Thanks for answering!

closed 2021-11-29 by xqrdot