carla-simulator/scenario_runner #852: 0.9.13 Scenario runner, error, ERROR: failed to destroy actor 722 : unable to destroy actor: not found

apiszcz, 2021-12-31:
Is this a common issue? How do I correct the error? Tx

```
 scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld --output
INFO:  Found the required file in cache!  Carla/Maps/TM/Town01.bin
Preparing scenario: FollowLeadingVehicle_1
ScenarioManager: Running scenario FollowVehicle

 ======= Results of Scenario: FollowVehicle ---- TIMEOUT =======

 > Ego vehicles:
Actor(id=722, type=vehicle.lincoln.mkz_2017);

 > Other actors:
Actor(id=723, type=vehicle.nissan.patrol);

 > Simulation Information
╒═════════════════════════════════╤═════════════════════╕
│ Start Time                      │ 2021-12-31 13:37:38 │
├─────────────────────────────────┼─────────────────────┤
│ End Time                        │ 2021-12-31 13:38:38 │
├─────────────────────────────────┼─────────────────────┤
│ Duration (System Time)          │ 60.14s              │
├─────────────────────────────────┼─────────────────────┤
│ Duration (Game Time)            │ 60.02s              │
├─────────────────────────────────┼─────────────────────┤
│ Ratio (System Time / Game Time) │ 0.998s              │
╘═════════════════════════════════╧═════════════════════╛

 > Criteria Information
╒═══════════════════════════╤══════════════════════╤═════════╤══════════════╤════════════════╕
│ Actor                     │ Criterion            │ Result  │ Actual Value │ Expected Value │
├───────────────────────────┼──────────────────────┼─────────┼──────────────┼────────────────┤
│ lincoln.mkz_2017 (id=722) │ CollisionTest (Req.) │ SUCCESS │ 0            │ 0              │
├───────────────────────────┼──────────────────────┼─────────┼──────────────┼────────────────┤
│                           │ Timeout (Req.)       │ FAILURE │ 60.02        │ 60             │
├───────────────────────────┼──────────────────────┼─────────┼──────────────┼────────────────┤
│                           │ GLOBAL RESULT        │ TIMEOUT │              │                │
╘═══════════════════════════╧══════════════════════╧═════════╧══════════════╧════════════════╛
 ===============================================================

Not all scenario tests were successful
Destroying ego vehicle 722
ERROR: failed to destroy actor 722 : unable to destroy actor: not found
No more scenarios .... Exiting
```

glopezdiest, 2022-01-10:
That's fairly common as we call several types the cleanup function, which tries to destroy the actor. However, that really isn't an ERROR, but a WARNING instead, as it doesn't stop the simulation, just prints out that message

closed 2022-01-10 by glopezdiest

Justin-Xiang, 2024-11-15:
I came across the same error, were you able to solve it?