carla-simulator/scenario_runner #853: OpenScenario Camera

steki555, 2022-01-07:
Hello! I would like to get the results of the camera attached to my ego_vehicle.

I have already added the following code to my xosc file.
``
[image]

How can I receive Camera Data and LIDAR Data to later evaluate? For example as JPG or PNG ?
Thank you in advance!

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.12]
 - Python version [e.g. 3.7]
 - Version [e.g. 0.9.12]

glopezdiest, 2022-01-10:
Hey @steki555. I've never trying using it but according to the code, [here](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenariomanager/actorcontrols/visualizer.py#L81) is the moment when the video is created and [here](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenariomanager/actorcontrols/visualizer.py#L130) is where each image is saved (per frame). Have you tried searching for that file?

steki555, 2022-01-10:
Hey @glopezdiest ! Thank you for your answer! I've tried searching for it, but unfortunately I don't have such files neither in scenario_runner nor somewhere by my xosc. files. Should i maybe have something else set in order to trigger those methods? Or maybe I should make my own object of the class Visualiser? Or maybe the XOSC ControllerAction should have some other value set to true?
 When i start the scenario, I  see the camera from ego_vehicle PoV and also from birds eye PoV.

glopezdiest, 2022-01-10:
Ah, apparently, that flag is always False. You can either change is [here](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenariomanager/actorcontrols/visualizer.py#L54), or modify the class to accept a new parameter.

@fabianoboril Do you have any more insights on this? I've never used this so I'm not sure what is its purpose. Can we add a new parameter to control this video writing?

steki555, 2022-01-11:
Thank you for your answer! It helped a lot and now I can also see the video after setting the flag to True. Do you maybe also have experience with LIDAR Data?

glopezdiest, 2022-01-11:
> Do you maybe also have experience with LIDAR Data?

What do you mean, visualizing the LIDAR as an image? If that's the case. I don't think there really is anything prepared for that in our examples. You can check [this function at the manual control](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/manual_control.py#L1132) that, each tick, parses the LIDAR data into an image (you only really care about the LIDAR conditional).

closed 2024-06-14 by glopezdiest