carla-simulator/scenario_runner #856: I tried to run manual_contro.py in scenario runner and I am getting this error "    from examples.manual_control import (World, HUD, KeyboardControl, CameraManager, ModuleNotFoundError: No module named 'examples'

umlautsim, 2022-01-21:
I am able to run "python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld" succesfully but when I try to run the manual_control I get this no module error. I have added the python paths as directed in the getting started page.

`export UE4_ROOT=~/UnrealEngine_4.26 
export CARLA_ROOT=/home/*username*/carla
export SCENARIO_RUNNER_ROOT=/home/*username*/scenario_runner
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-0.9.12-py3.6-linux-x86_64.egg
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/agents`

Kron610, 2022-01-26:
Try to add "export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/examples"

glopezdiest, 2022-01-28:
As a follow up to what Kron610 said, we recently removed that dependency ([this is the commit](https://github.com/carla-simulator/scenario_runner/commit/06d912b78a34522795b7c2b290c693e34b3ba6a4), as it wasn't really a good idea to begin with.

closed 2022-01-28 by glopezdiest

LucasFKobernic, 2022-02-14:
Hi @glopezdiest 

I'm facing the same issue. I am on windows and after a added:

set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI\examples\

The error persists. Any idea what is causing the issue? I'm usgin Scenario Runner 0.9.12

umlautsim, 2022-06-13:
HI @LucasFKobernic, I am facing the same error. Did you find any fix?. Thanks