carla-simulator/scenario_runner #872: Ego vehicle speed with manual_control.py

joluko, 2022-03-22:
**Describe our question or idea**
Hello,
I am trying to assign an initial speed to the ego vehicle while using the carla autopilot in manual_control.py. I am defining the scenarios in OpenSCENARIO files. When I run the scenario  the ego vehicle accelerates at first but then brakes abruptly, this behavior repeats itself (see figure).
So I used the function global_percentage_speed_difference(-100) to 'ignore' the speed limit. 

Can someone explain this behavior to me? Why does the ego vehicle brake and how can i avoid that? 

I am glad about any help or input on this.

**Expected behavior**
I expect the vehicle to hold the given speed until a leading vehicle or object appears.

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - CARLA 0.9.11
 - Python version 3.7.11
 - Version 0.9.11

glopezdiest, 2022-03-30:
That's probably because you are running the simulation asynchronously, which caused the autopilot's PID to not work properly. Make sure to switch to synchronous mode, and the autopilot will work as intended.

Georgezxh, 2022-05-07:
> That's probably because you are running the simulation asynchronously, which caused the autopilot's PID to not work properly. Make sure to switch to synchronous mode, and the autopilot will work as intended.

Hi, I wonder that why the autopilot mode in manual_control can't work when I enter P in keyboard.

glopezdiest, 2022-05-26:
@Georgezxh What do you mean by *can't work*?