carla-simulator/scenario_runner #875: Possible Bug: Ego vehicle not able to make use of SpeedActionDynamics when under simple_vehicle_control using openSCENARIO scenario generation

werewolfdev, 2022-03-28:
Hi team, Iam trying to create scenarios using openSCENARIO scripts. I am trying to create a simple scenario where the ego_ vehicle is controlled by simple_vehicle_control and made to move from 0kmph to 20 kmph in with SpeedDynamics linear . I also tried with other Shapes . However all the time the speed value just shoots up from 0 to 20kmph . I would like to have a linear or stepwise transition for the speed parameter of my ego vehicle. Could anyone let me know what I myt be  missing out : The respective SpeedAction code segment :
<SpeedAction>
                      <SpeedActionDynamics dynamicsShape="linear" value="50" dynamicsDimension="distance"/>
                      <SpeedActionTarget>
                        <AbsoluteTargetSpeed value="5.0"/>
                      </SpeedActionTarget>
                    </SpeedAction>
It works for actors run as simulation but not for ego_vehicle

 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.13]
 - Python version [e.g. 3.6]
 - Version [e.g. 0.9.13]

werewolfdev, 2022-03-29:
@fabianoboril  Hi Could you please provide your insights on the same. Thanks

glopezdiest, 2022-03-30:
Hey @werewolfdev. Your SpeedAction code seems fine, could provide the whole script you are using to reproduce the erros? As well as how are you calling scenario runner.

werewolfdev, 2022-03-30:
Hey @glopezdiest  thanks a lot for the reply. The xosc script is as below.  Iam running the scenario using scenario_runner script normally from the scenario runner folder of CARLA repository liek below:
python3.6 scenario_runner.py --openscenario /home/schemba/CARLA_0.9.13/carla/scenario_runner-0.9.13/srunner/examples/Abbiege_FRTest.xosc

`<?xml version="1.0" ?>
<OpenSCENARIO>
    <FileHeader author="QGIS OSCGenerator Plugin" date="2021-09-22T14:40:12" description="Generated OpenSCENARIO File" revMajor="1" revMinor="0"/>
    <ParameterDeclarations/>
    <CatalogLocations/>
    <RoadNetwork>
        <LogicFile filepath="Town01"/>
        <SceneGraphFile filepath=""/>
    </RoadNetwork>
    <Entities>
        <ScenarioObject name="hero">
            <Vehicle name="vehicle.tesla.model3" vehicleCategory="car">
                <ParameterDeclarations/>
                <Performance maxAcceleration="200" maxDeceleration="10.0" maxSpeed="69.444"/>
                <BoundingBox>
                    <Center x="1.5" y="0.0" z="0.9"/>
                    <Dimensions height="1.8" length="4.5" width="2.1"/>
                </BoundingBox>
                <Axles>
                    <FrontAxle maxSteering="0.5" positionX="3.1" positionZ="0.3" trackWidth="1.8" wheelDiameter="0.6"/>
                    <RearAxle maxSteering="0.0" positionX="0.0" positionZ="0.3" trackWidth="1.8" wheelDiameter="0.6"/>
                </Axles>
                <Properties>
                    <Property name="type" value="ego_vehicle"/>
                </Properties>
            </Vehicle>
        </ScenarioObject>
        <ScenarioObject name="Vehicle_1">
            <Vehicle name="vehicle.audi.tt" vehicleCategory="car">
                <ParameterDeclarations/>
                <Performance maxAcceleration="200" maxDeceleration="10.0" maxSpeed="69.444"/>
                <BoundingBox>
                    <Center x="1.5" y="0.0" z="0.9"/>
                    <Dimensions height="1.8" length="4.5" width="2.1"/>
                </BoundingBox>
                <Axles>
                    <FrontAxle maxSteering="0.5" positionX="3.1" positionZ="0.3" trackWidth="1.8" wheelDiameter="0.6"/>
                    <RearAxle maxSteering="0.0" positionX="0.0" positionZ="0.3" trackWidth="1.8" wheelDiameter="0.6"/>
                </Axles>
                <Properties>
                    <Property name="type" value="simulation"/>
                </Properties>
            </Vehicle>
        </ScenarioObject>
    </Entities>
    <Storyboard>
        <Init>
            <Actions>
                <GlobalAction>
                    <EnvironmentAction>
                        <Environment name="Environment1">
                            <TimeOfDay animation="false" dateTime="2020-10-23T04:00:00"/>
                            <Weather cloudState="free">
                                <Sun azimuth="0" elevation="1.31" intensity="0.85"/>
                                <Fog visualRange="100000"/>
                                <Precipitation intensity="0" precipitationType="dry"/>
                            </Weather>
                            <RoadCondition frictionScaleFactor="1.0"/>
                        </Environment>
                    </EnvironmentAction>
                </GlobalAction>
                <Private entityRef="hero">
                    <PrivateAction>
                        <TeleportAction>
                            <Position>

                                <WorldPosition h="1.5709483394834305" p = "0" r = "0" x="92.32854315481438" y="-135.08543140104072" z="0.2"/>
[... 284 lines cut ...]
                    </ConditionGroup>
                </StartTrigger>
                <StopTrigger>
                    <ConditionGroup>
                        <Condition conditionEdge="rising" delay="0" name="EndCondition">
                            <ByValueCondition>
                                <SimulationTimeCondition rule="equalTo" value="10000"/>
                            </ByValueCondition>
                        </Condition>
                    </ConditionGroup>
                </StopTrigger>
            </Act>
        </Story>
        <StopTrigger>
            <ConditionGroup>
                <Condition name="criteria_DrivenDistanceTest" delay="0" conditionEdge="rising">
                    <ByValueCondition>
                        <ParameterCondition parameterRef="distance_success" value="100" rule="lessThan"/>
                    </ByValueCondition>
                </Condition>
            </ConditionGroup>
        </StopTrigger>
    </Storyboard>
</OpenSCENARIO>
`

glopezdiest, 2022-03-30:
Great, I'll test it ASAP. Feel free to remind me if I haven't said anything by next week

werewolfdev, 2022-03-30:
Thanks .sounds good .Yeah surely will remind you 😂😂😂

glopezdiest, 2022-04-01:
> It works for actors run as simulation but not for ego_vehicle
 
Hey @werewolfdev, from what I can see, neither the ego nor the other actor does change speed linearly. The issue you are seeing is caused by a series of factors.

First of all, when you run the *.xosc*, we get the xml file and parse it with the [OpenScenarioParser](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/tools/openscenario_parser.py). For the `SpeedAction` specifically, this is done [here](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/tools/openscenario_parser.py#L1175), and as you can see, **we are not parsing the dynamic shape**, so ScenarioRunner doesn't know that's what you want.

From there, we call the [ChangeActorTargetSpeed](https://github.com/carla-simulator/scenario_runner/blob/f4d00d88eda4212a1e119515c96281a4be5c234e/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py#L408), which tells the vehicle's controller that the target speed has changed (unexpected, I know). In your case, you are using the [SimpleVehicleControl](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenariomanager/actorcontrols/simple_vehicle_control.py), which moves the vehicle not with physics, but by forcing its speed ([here](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenariomanager/actorcontrols/simple_vehicle_control.py#L309)). As such, it is capable of instantly changing the speed, so that's what it does.

As for the solution, it's a bit tricky. We should definitely add the parsing of the dynamic shape. In the meantine, you can check the [NPCVehicleControl](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenariomanager/actorcontrols/npc_vehicle_control.py), which actually uses physics to move the vehicle, so the linear shape will be done. BUT, the step won't (as it is now bound by physics) so you'll have to mix the two controllers together.

werewolfdev, 2022-04-01:
Hey @glopezdiest . Thanks a lot for the reply. Hope the community could add this feature soon. Regarding the solution , I can change the type of module to npc_vehicle_control for the ego vehicle: Actually I need just the linear speed changes and not the step dynamics now, to generate the required scenarios. So as per your views, just changing the type of control to npc_vehicle_control will do the trick for me. I will try it out and let you know son. Please correct me if Iam wrong

glopezdiest, 2022-04-01:
Correct, you need:
- NPCVehicleControl for the the linear shape
- SimpleVehicleControl for the the step shape
If you just need one shape, just use that specific controller

werewolfdev, 2022-04-01:
@glopezdiest Thanks. So i have change controller as npc_vehicle_control in xosc script but then when i run it , no window opens like simple_vehicle_control. In npc_vehicle_control is there any additional code I should run to run the simulation or everything same as in simple_vehicle_control. I would actually like to run the ego vehicle also like simulation and not from key pad. Thanks

glopezdiest, 2022-04-01:
Oh yeah that's true, the simple_vehicle_control does open a window, which other controllers don't. You'd have to integrate it :slightly_smiling_face:. 

>  I would actually like to run the ego vehicle also like simulation and not from key pad

Can't you use the npc_vehicle_control with the ego to not move it with the key pad?

werewolfdev, 2022-04-01:
oh is it. Yeah Iam using npc . I hope i just need to integrate that window opening .Ok i would look into the simple_vehicle_control script and see how to integrate this. I hope thats where I have to look into. I will update you soon

werewolfdev, 2022-04-03:
Hi @glopezdiest I integrated visualizer into npc vehicle control script and tested the control action of linear SpeedActionDynamics. It works. Thanks a lot for your help.

closed 2022-04-03 by werewolfdev