carla-simulator/scenario_runner #876: How to save some data from ScenarioRunner runtime scenarios?

GeneralMak, 2022-03-29:
How can I save the data from ScenarioRunner runtime scenarios?  Parameters such as speed, collsion etc.
[image]
[image]
Thanks in advance! @glopezdiest

glopezdiest, 2022-03-30:
There isn't really an exact way to do it, as it heavily depends on the output format you are expecting. You can output the criteria results by using the _--file_, _--junit_ or _--json_ arguments to output the data in specific formats. For routes (as that is what you are showignin the image), [here](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenarios/route_scenario.py#L448) are the criteria we use.

You can also use the [metrics module](https://carla-scenariorunner.readthedocs.io/en/latest/metrics_module/) to create your own metrics, which will be outputted in your own custom format.

glopezdiest, 2022-04-01:
So you can't use the metrics manager with the data recorded by a route? It just gives you a client timeout. This might be because of the amount of data, yeah. Try to increase the timeout of the client to more than 5 seconds

GeneralMak, 2022-04-02:
Thanks for the reply, I tried setting the client timeout  for longer than 5 seconds, but it doesn't seem to work. Still getting **_RuntimeError:time-out of 5000ms_**

glopezdiest, 2022-05-26:
Hey @GeneralMak, so sorry for the delay, it completely slipped my mind. Did you manage to make it work for your use-case?

This might not be relevant anymore but routes are just more complex scenarios with subscenarios inside it, so yes, you can use the metrics manager at routes.

olyii, 2022-12-11:
> Thanks for the reply, I tried setting the client timeout for longer than 5 seconds, but it doesn't seem to work. Still getting **_RuntimeError:time-out of 5000ms_** Can you use **_metrics_manager_** to measure route scenarios?

Have you save some data from ScenarioRunner runtime scenarios successfully?I want to know how to make it ,thank you