carla-simulator/scenario_runner #879: How to change the number of the npc while running with route?

Watson52, 2022-04-14:
Hi :) 
I am using the route model to run the SR. I found that there has set town_amount in the **background_activity.py** file but when I try to reduce the size of the amout it didn't work. The  number of 'background' actors still did not change. Did I miss anything?
Looking forward to your reply.

 - OS:  Ubuntu 18.04
 - CARLA Version 0.9.10
 - Python version 3.7
 - Version 0.9.10

Watson52, 2022-04-15:
Oh, it can be change in the route_scenario.py file. I just missed it.

closed 2022-04-15 by Watson52

shh1v, 2023-07-26:
Just for future readers, please also look into `background_activity.py`. By default, `route_scenario.py` spawns 0 background actors unless changed. So, modifying `background_activity.py` is probably what you want.