carla-simulator/scenario_runner #881: use Private FollowTrajectory Actions with CARLA and ScenarioRunner 0.9.13

jch458, 2022-04-20:
**Describe our question or idea**
Dear CARLA team,

I am new to CARLA and OSC 1.0, and I am trying to generate some simple OSC files to be visualized in CARLA. But I have problems in using the Private FollowTrajectory Actions, and got this error: 
```
NotImplementedError: Private FollowTrajectory actions are not yet supported
Private FollowTrajectory actions are not yet supported
```

The document suggests that the latest version of CARLA supports Private Action -> RoutingAction -> FollowTrajectoryAction in the Story block, with commenting "Currently only Polylines without temporal constraints are supported and the TrajectoryFollowingMode is up to the controller." (https://carla-scenariorunner.readthedocs.io/en/latest/openscenario_support/) I am not sure if my following OSC code is valid to version 0.9.13 --- and if not, what a valid TrajectoryFollow action should be like. Could anyone please have a look? Thank you.

```
<PrivateAction>
  <RoutingAction>
    <FollowTrajectoryAction>
      <Trajectory closed="false" name="">
        <Shape>
          <Polyline>
            <Vertex time="0">
                <Position>
                  <RelativeRoadPosition entityRef="ego" ds="60" dt="-1.0"/>
                </Position>
            </Vertex>
            <Vertex time="0">
                <Position>
                  <RelativeRoadPosition entityRef="ego" ds="70" dt="-1.0"/>
                </Position>
            </Vertex>
          </Polyline>
        </Shape>
      </Trajectory>
      <TimeReference />
      <TrajectoryFollowingMode followingMode="follow" />
    </FollowTrajectoryAction>
  </RoutingAction>
</PrivateAction>
```

**Desktop (please complete the following information):**
 - OS: [e.g. Win 10]
 - CARLA Version [e.g. CARLA 0.9.13]
 - Python version [e.g. 3.7]
 - Version [e.g. 0.9.13]

jch458, 2022-04-22:
Dear all, I use the master branch of SR and get the feature of the follow trajectory action. Thanks!

closed 2022-04-22 by jch458

vibhoraggarwal, 2022-07-08:
Hi @jch458 , I am using the followtrajectory option as well with the latest master branch and Carla 0.9.13. The .xosc file does get executed fine but in the end it showed the following error. Did you also receive it, or have any idea to solve it?

### Error

```
File "D:\git\scenario_runner_carla_git\srunner\scenariomanager\timer.py", line 122, in update
    if not self._comparison_operator(elapsed_time, self._timeout_value):
  File "D:\git\scenario_runner_carla_git\srunner\tools\openscenario_parser.py", line 199, in __lt__
    return self.__float__() < other
  File "D:\git\scenario_runner_carla_git\srunner\tools\openscenario_parser.py", line 142, in __float__
    raise Exception("could not convert '{}' to float".format(self.reference_text))
Exception: could not convert '10e15' to float
could not convert '10e15' to float
```

### Additional infrormation:

    OS: Win 10
    CARLA Version: CARLA 0.9.13 (pre-built)
    Python version: 3.7.10
    ScenarioRunner Version: 0.9.13