carla-simulator/scenario_runner #896: ModuleNotFoundError: No module named 'agents'

chenhengwei1999, 2022-05-18:
I want to use scenario runner in Carla simulator. I follow the tutorial on the official website(Calra-Simulator:https://carla.readthedocs.io/en/latest/start_quickstart/ ,Scenario-Runner:https://carla-scenariorunner.readthedocs.io/en/latest/getting_scenariorunner/) and have installed package version carla-0.9.13. Then I need to find the corresponding version of scenario runner.

But I can't find the corresponding non-source download package in scenario, as shown in the figure below. I can only find the source version(This will conflict with my package version of Carla). 
[image]

So I can only download the source version of scenario runner(*.tar.gz), but the following errors occur at runtime.
[image]

In order to check the errors, I tried to import the agents package in conda environment, but the same error occurred. Does Carla of 0.9.13 package need to download the agents package separately to run scenario runner? as shown in the figure below
[image]

How do I check for errors? Thank you in advance

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - CARLA Version [e.g. CARLA 0.9.13]
 - Python version [e.g. 3.7
 - Scenario Runner Version [e.g. 0.9.6]

vimal4353, 2022-05-19:
Please set the carla agents folder path in atomic_behavior.py file

chenhengwei1999, 2022-05-19:
> Please install pip install agents and set the carla agents folder path in atomic_behavior.py file

Thanks for you reply. I tried add the PYTHONPATH=.../path/to/carla/root/PythonAPI/carla into the ~/.bashrc file.
So the <agents> python thrid library can be imported successfully.

By the way, I could't find the location where atomic_behavior.py exits. What is the functionality of the file? If possible, could you please give me some details about it, such as why i need to use it(I am a new Carla-er, and i am not familar with the simulator, please give me more patience, THANKS).

I am using the carla-0.9.13 and ubuntu 20.04.

vimal4353, 2022-05-19:
You can set the path in C:\Users\Admin\carla\PythonAPI\scenario_runner-0.9.11\srunner\scenariomanager\scenarioatomics\atomic_behaviors.py

sys.path.insert(0,'C:/Users/Admin/carla/PythonAPI/carla')    -   put before importing the agents module

chenhengwei1999, 2022-05-19:
> You can set the path in C:\Users\Admin\carla\PythonAPI\scenario_runner-0.9.11\srunner\scenariomanager\scenarioatomics\atomic_behaviors.py
> 
> sys.path.insert(0,'C:/Users/Admin/carla/PythonAPI/carla') - put before importing the agents module

Thank you very much for your advice.

glopezdiest, 2022-05-26:
@chenhengwei1999 Others have said it already, but the agents folder has to be added to the PYTHONPATH in order to be used, as seen [here](https://carla-scenariorunner.readthedocs.io/en/latest/getting_scenariorunner/#update-from-source), from there, just import the agents folder in python ([an example](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py#L33)).

Closing this issue as it seems resolved, but feel free to reopen it if you still have issues

closed 2022-05-26 by glopezdiest