carla-simulator/scenario_runner #897: AttributeError: Only OpenSCENARIO 1.0 is supported

vimal4353, 2022-05-19:
Hi, I am using carla 0.9.11 and tried with my own custom .xosc file. My xosc file is created in openscenario 1.1 standard.. please let me know how to solve this attribute error.

glopezdiest, 2022-05-26:
The error you are seeing is [here](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenarioconfigs/openscenario_configuration.py#L96). Either remove the check or change your version to 1.0

closed 2022-05-26 by glopezdiest

vimal4353, 2022-05-31:
How to remove the check ? Is there any possibility to run the OSC 1.1 in carla simulation.

vimal4353, 2022-05-31:
> The error you are seeing is [here](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenarioconfigs/openscenario_configuration.py#L96). Either remove the check or change your version to 1.0

Please let me know Is there any possibility to run the OSC 1.1 in carla simulation.