carla-simulator/scenario_runner #899: Remove Traffic Manager from scenario_runner.py

VinalAsodia, 2022-05-21:
Hi, I am trying to run a DynamicObjectCrossing route-based scenario with no surrounding vehicles in the world. From reading the online documentation and looking through the scenario_runner.py file, it's the Traffic Manager that generates the background traffic. I have tried to comment out the instantiation and setup of the Traffic Manager, as well as its cleanup, but surrounding vehicles still spawn when the route-based scenario is executed. I would appreciate any help on this!

Setup:
 - OS: Ubuntu 20.04
 - CARLA Version: CARLA 0.9.13
 - Python version: 3.8.5
 - Scenario Runner Version: 0.9.13

VinalAsodia, 2022-05-23:
Never mind, I managed to solve this issue. I looked at the RouteScenario class that all route-based scenarios inherit from and found a method called "_initialize_actors" and once removing the call of this method, the other vehicles in the town were removed!

closed 2022-05-23 by VinalAsodia

FBanani, 2022-08-03:
> Never mind, I managed to solve this issue. I looked at the RouteScenario class that all route-based scenarios inherit from and found a method called "_initialize_actors" and once removing the call of this method, the other vehicles in the town were removed!

Hi, can I allocate StationaryObjectCrossing class to vehicles that are provided by generate_traffic.py?