carla-simulator/scenario_runner #907: Why is there limitation at the steering when using waypoint to control vehicle in scenario_runner?

holymoly-JihyeSeo, 2022-06-16:
I'm trying to create a scenario for a vehicle to turn right or left at the intersection using waypoint.  
So i had to understand the code at atomic behaviors.py (https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py) and especially the class ChangeActorWaypoints.

However, the scenario does not work as expected because of the code below:

```
class ChangeActorWaypoints(AtomicBehavior):

      ,,,

        def initialise(self):

     ,,, 
      
                        heading_delta = math.fabs(new_heading - last_heading)
                        if math.fabs(heading_delta) < 0.5 or math.fabs(heading_delta) > 5.5:
                            route.append(wp_tuple[0].transform)

       '''
```
            

The scenario i created using Waypoint gives the value of heading_delta between 0.5 and 5.5. Thus, it is not added to the array `route`. 

Why is there limitation to the range of `heading_delta`? and what does the value 0.5 and 5.5 means??

glopezdiest, 2022-08-05:
Hello @holymoly-JihyeSeo, from comments inside the code:

```py
# The router sometimes produces points that go backward, or are almost identical
# We have to filter these, to avoid problems
```
What you checked that this is not the case for your waypoints?

ronyshaji, 2023-08-01:
@holymoly-JihyeSeo Hey did you find a solution for this issue?