carla-simulator/scenario_runner #919: I tried changing the path and everything seems correct but this error keeps coming :

laks1806, 2022-07-12:
@fabianoboril, I tried changing the path and everything seems correct but this error keeps coming : 
```
python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld
Configuration for scenario FollowLeadingVehicle_1 cannot be found!
No more scenarios .... Exiting
```
my python path is: 
`/opt/carla-simulator/PythonAPI/carla/dist/carla-0.9.11-py2.7-linux-x86_64.egg:/opt/carla-simulator/PythonAPI/carla/agents:/opt/carla-simulator/PythonAPI/carla`

_Originally posted by @laks1806 in https://github.com/carla-simulator/scenario_runner/issues/311#issuecomment-1180375753_

wkgud9090, 2022-07-26:
have same problem

glopezdiest, 2022-08-05:
As @ghazalmatt3r said, you should have an environment variable pointin to your Scenario runner folder for the configurations to be found. Closing the issue as it is resolved, but feel free to reopen it if needed

closed 2022-08-05 by glopezdiest