carla-simulator/scenario_runner #921: IndexError: deque index out of range

vimal4353, 2022-07-18:
Hi all,
I tried scenario runner 0.9.13 in Carla 0.9.13 version.
I am getting these error, How to fix this please let me know.

Traceback (most recent call last):
  File "scenario_runner.py", line 413, in _load_and_run_scenario
    self.manager.run_scenario()
  File "C:\Users\Admin\carla\PythonAPI\scenario_runner-0.9.13-new\scenario_runner-0.9.13\srunner\scenariomanager\scenario_manager.py", line 139, in run_scenario
    self._tick_scenario(timestamp)
  File "C:\Users\Admin\carla\PythonAPI\scenario_runner-0.9.13-new\scenario_runner-0.9.13\srunner\scenariomanager\scenario_manager.py", line 178, in _tick_scenario
    self.scenario_tree.tick_once()
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python37\lib\site-packages\py_trees\behaviour.py", line 158, in tick_once
    for unused in self.tick():
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python37\lib\site-packages\py_trees\composites.py", line 578, in tick
    for node in child.tick():
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python37\lib\site-packages\py_trees\behaviour.py", line 249, in tick
    new_status = self.update()
  File "C:\Users\Admin\carla\PythonAPI\scenario_runner-0.9.13-new\scenario_runner-0.9.13\srunner\scenariomanager\scenarioatomics\atomic_behaviors.py", line 406, in update
    actor_dict[actor_id].run_step()
  File "C:\Users\Admin\carla\PythonAPI\scenario_runner-0.9.13-new\scenario_runner-0.9.13\srunner\scenariomanager\actorcontrols\actor_control.py", line 180, in run_step
    self.control_instance.run_step()
  File "C:\Users\Admin\carla\PythonAPI\scenario_runner-0.9.13-new\scenario_runner-0.9.13\srunner\scenariomanager\actorcontrols\npc_vehicle_control.py", line 108, in run_step
    control = self._local_planner.run_step(debug=False)
  File "C:/Users/Admin/carla/PythonAPI/carla\agents\navigation\local_planner.py", line 234, in run_step
    self._compute_next_waypoints(k=100)
  File "C:/Users/Admin/carla/PythonAPI/carla\agents\navigation\local_planner.py", line 178, in _compute_next_waypoints
    last_waypoint = self._waypoints_queue[-1][0]
IndexError: deque index out of range
deque index out of range
Resetting ego-vehicle!

vimal4353, 2022-07-22:
Hi all,
I solved this issue by giving a reach position condition in the inside story tag.

closed 2022-07-22 by vimal4353

bapukokare, 2022-12-02:
waypoints are automatically fetched with road id information and follow the same lane - roadoption= lane follow set in the local planner, if you want to follow a compulsory trajectory what to do?