carla-simulator/scenario_runner #932: Modifying maximal velocity of hero controlled by autonomous agent using .xosc file

openscenario, 2022-08-25:
I need to modify the maximal speed the autonomous agent is driving in my scenario. In my scenario.xosc file I tried to modify the maxSpeed value in the ScenarioObject definition, but this does not affect the maximal speed.
```
...
<Entities>
    <ScenarioObject name="hero">
      <Vehicle name="vehicle.volkswagen.t2" vehicleCategory="car">
        <ParameterDeclarations/>
        <Performance maxSpeed="2" maxAcceleration="80" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties>
          <Property name="type" value="ego_vehicle"/>
        </Properties>
      </Vehicle>
    </ScenarioObject>
...
```

I assume setting an initial speed action does not makes sense, since the hero behaviour is controlled externally. 

I have just encountered that the BasigAgent init function has target_speed parameter: https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/basic_agent.py#L29

Is there a way of setting this parameter using the scenario.xosc file, or is there something I am not aware of?

TigerStone93, 2022-10-27:
Is there any update on here?

Now the `Performance` property initialize the `maxSpeed` on its `Vehicle` well?

Leviathan321, 2022-10-27:
Yes I can confirm. I could pass the velocity by passing the max velocity value to the BasicAgent seehttps://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/basic_agent.py. 

Parameters as maxAcceleration and maxDeceleration, cannot be passed through the agent as it looks like.