carla-simulator/scenario_runner #937: Openscenario standard v2.0

VigneshMohan123, 2022-09-21:
Is it possible to execute scenarios in open scenario v2.0 or any open scenario standard higher than v1.0? I noticed in the documentation for Carla that the scenario runner only works with version 1.0!
[image]

glopezdiest, 2022-09-22:
> Is it possible to execute scenarios in open scenario v2.0 or any open scenario standard higher than v1.0?

Technically no, we currently only partially support OSC 1.0, so any changes done by higher versions are not supported.

However, you can always do it yourself, and while that's probably not a satisfactory answer, if you are interested in a specific feature, I can provide some help with guidelines on how the OSC 1.0 support is done

MaxPBauer, 2022-10-05:
The difference between OpenSCENARIO 1.x and 2.0 is extreme and the new standard is overloaded with a lot of new concepts. Implementing Osc2.0 is extremely hard and requires IMO joint effort to make an open source implementation.

For your information, I am the main developer behind [Yase](https://gitlab.eclipse.org/eclipse/simopenpass/yase) and we intend to do exactly this. The open sourced Yase framework is build to support multiple scenario formats (such as in [OpenPASS with Osc1.x](https://gitlab.eclipse.org/eclipse/simopenpass/openscenario1_engine) or company internal formats). This can be implemented **agnostic** of the simulator, so we can have a reuse of the frontend for many simulators.

Internally we already use an Osc2 frontend with our internal simulators, however we still relay on some internal code. Once it is [fully removed](https://gitlab.eclipse.org/eclipse/simopenpass/yase/-/issues/23), we will share it so it can be used for more simulators such as carla. Until then one can already look into some (outdated) [code](https://gitlab.eclipse.org/eclipse/simopenpass/yase/-/merge_requests/6) to get the rough idea. 

If you plan to do something in direction Osc2.0, feel free to contact us so we can see how we can support & avoid redundant code!

ronyshaji, 2023-09-01:
@SEEKERXC I am having an issue with OpenScenario while using the waypoint based scenario. Apparently the scenario runner is stuck with first event and not going to the next event. I used StoryboardElementStateCondition as the start trigger and do you have any experience or feedback about this ?

jasmeetgaba24, 2023-10-25:
@XiaoFei9704 , I am facing some challenges with Path Modifiers in Open Scenario 2.0, Can you provide some insights on same.
Issue explained here - https://github.com/carla-simulator/scenario_runner/issues/1027