carla-simulator/scenario_runner #953: No module named 'agents.navigation.constant_velocity_agent' for leaderboard-2

SiyuanHuang95, 2022-11-25:
**Describe our question or idea**

A clear and concise description.
When I am trying to use leaderboard.py to do some evaluations, the import error occurs as follows:
```
 No module named 'agents.navigation.constant_velocity_agent'
```
The only exception between my execution and tutorials is that I used carla0.9.13  not the carla-leaderboard2. Was that the cause?

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 16.04] Ubuntu 20.04
 - CARLA Version [e.g. CARLA 0.9.6] 0.9.13
 - Python version [e.g. 2.7] 3.7
 - Version [e.g. 0.9.6] 0.9.13 - -Leaderboard2

SiyuanHuang95, 2022-12-10:
> Yes, as you mentioned you should use: https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/Leaderboard/CARLA_Leaderboard_20.tar.gz the carla leaderboard release. Otherwise it won't work.
> 
> edit: The first response was slightly wrong.

Correct, after switching to that release, the error disappeared. But other errors appeared, LOL. It seems Leaderboard2 is still under development.

haixuanTao, 2022-12-11:
Mind sharing what other `errors` did you faced?

AOOOOOA, 2023-05-02:
> Correct, after switching to that release, the error disappeared. But other errors appeared, LOL. It seems Leaderboard2 is still under development.

Hi have you solved the problem?

Ezharjan, 2023-12-23:
> **Describe our question or idea**
>
> A clear and concise description. When I am trying to use leaderboard.py to do some evaluations, the import error occurs as follows:
>
> ```
>  No module named [...]

The major cause of this problem may be:
1. **Version mismatch** between ScenarioRunner and CARLA
2. `CARLA-agents` directory may not be exported properly.

Here is a series of commands to restart from the first step as a **solution** to fix your problem regarding the causes above:
```bash
# Download CARLA with the matching version of scenario_runner, eg: carla0.9.13 for scenario_runner9.13
conda create srunner python=3.7
conda activate srunner
pip3 install --user -r requirements.txt
pip install carla==0.9.13 # carla version has to match the version with scenario_runner
pip uninstall tensorflow
pip install tensorflow==1.15.0
pip install agents
pip install pygame

# Run the following commands everytime you open a new terminal
unset $PYTHONPATH # this will unset the mistakenly set `PYTHONPATH` via multiple run of the `export` command.
unset $CARLA_ROOT # this will unset the mistakenly set `CARLA_ROOT` via multiple run of the `export` command.
export CARLA_ROOT=/home/alex/Documents/Carla913 # Change the path to your own!!! DO NOT RUN THIS COMMAND MORE THAN ONCE, or you have to run `unset` mentioned above.
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-0.9.13-py3.7-linux-x86_64.egg:${CARLA_ROOT}/PythonAPI/carla/agents:${CARLA_ROOT}/PythonAPI/carla # Change the path to your own!!! DO NOT RUN THIS COMMAND MORE THAN ONCE, or you have to run `unset` mentioned above.

# Navigate to another terminal
./CarlaUE4.sh -carla-port=2000 # run CARLA in another terminal with port:2000

# Back to the original terminal
python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld --output # run the demo
```