carla-simulator/scenario_runner #971: I'm unable to run scenario-runner on windows, here is the screenshot of the error.

BhusareHarshad, 2023-03-10:
**Describe our question or idea**
I'm trying to run a simple default scenario on windows using scenario runner, but I'm unable to do so. I have attached the error screenshot below.
I have directly took the CARLA and Scenario Runner zip folders of same version and extracted to C drive.
I'm running CARLAUE4.exe and then scenario_runner.py file from CMD and have installed all the requirements from requirements.txt.
I have also set the PATH VARIABLES and below.

###############
set CARLA_ROOT=C:\CARLA_scenario\CARLA_0.9.13\WindowsNoEditor

set SCENARIO_RUNNER_ROOT=C:\CARLA_scenario\CARLA_0.9.13\WindowsNoEditor\scenario_runner-0.9.13\scenario_runner-0.9.13

set PYTHONPATH=C:\\Users\\usr1\\AppData\\Local\\Programs\\Python\\Python36;C:\CARLA_scenario\CARLA_0.9.13\WindowsNoEditor\PythonAPI\carla\dist\carla-0.9.13.egg

set PYTHONPATH=C:\\Users\\usr1\\AppData\\Local\\Programs\\Python\\Python36;C:\CARLA_scenario\CARLA_0.9.13\WindowsNoEditor\PythonAPI\carla\agents

set PYTHONPATH=C:\\Users\\usr1\\AppData\\Local\\Programs\\Python\\Python36;C:\CARLA_scenario\CARLA_0.9.13\WindowsNoEditor\PythonAPI\carla\

set PYTHONPATH=C:\\Users\\usr1\\AppData\\Local\\Programs\\Python\\Python36;C:\CARLA_scenario\CARLA_0.9.13\WindowsNoEditor\PythonAPI\

###############

**Desktop (please complete the following information):**
 - OS: Windows 11
 - CARLA 0.9.13
 - Python 3.6.5

 **Screenshots**
[image]

onebitme, 2023-04-19:
Please install `requirements.txt` via using 
`pip install -r requirements.txt` in scenario runner repo folder.
Also make sure that the windows command line python and the environment you are installing via pip are the same. (if your pip, for eg installs to Python 3.10 but you are using 3.7 for scenario runner, it will also cause a problem"

starrye, 2023-10-13:
Run python in your terminal, import xmlschema, and if it fails, install xmlschema using your package manager