carla-simulator/scenario_runner #972: how to save data from scenario runner?

fatemehprhm, 2023-03-11:
Hi, i want to run followleadingvehicle scenario and save and print the data in the terminal, i need speed and location and also the distance between two vehicles, is it possible to write a code that retrieves this data from scenario runner? i mean if i write this code do i have to run it seperately from scenario runner? or can i edit the the .py file of the scenario in a way that it prints this data in terminal? i know that its possible to record data but i want sth like distance that i should compute it myself.
thank you

Anhar2, 2023-06-05:
Hello,

Have you found a solution for that?

fatemehprhm, 2023-08-01:
@Anhar2 
Hi, sorry for my late reply. Yes i used manual_control.py to write my own data extractor. i  just added some more lines to get the location and other data, then computed distance and saved them in a csv file. By using  "print" it can be shown on the terminal too. Let me know if you need more information about the approach.

closed 2023-11-05 by fatemehprhm