carla-simulator/scenario_runner #973: Save sensor data from scenario runner

Dan-A22, 2023-03-13:
Hi everybody. I've been working with carla for half a year now, but It has been a few days that I'm working with scenario runner. I want to attach some cameras and radar to the ego vehicle and save the data for further analysis. Does anyone know where exactly can I write the code to do this?
I already know how to work with carla sensors and record the data. However I don't know where to implement this in scenario runner.
I appreciate your help!

**Desktop:**
 - OS: [Windows 10]
 - CARLA Version [CARLA 0.9.13]
 - Python version [37]
 - Version [0.9.13]

fatemehprhm, 2023-03-28:
Hi Dan, i have the same problem, could find any solution to this problem?

Anhar2, 2024-03-22:
Hi

There is the recorder feature on the scenario runner that works just fine, could you have any other approach though? 

@fatemehprhm