carla-simulator/scenario_runner #975: I cannot start Scenario_Runner by Carla.

GhaithNasleh, 2023-03-21:
Hello,

I am trying to get Scenario Runner by Carla to run. 

I use windose and have done all the required steps exactly. 

https://github.com/carla-simulator/scenario_runner/blob/master/Docs/getting_scenariorunner.md

I have also installed all the requirements. 

Unfortunately I get this error message: 

D:\scenario_runner>python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld

RuntimeError: module compiled against API version 0xe but this version of numpy is 0xd
ImportError: numpy.core.multiarray failed to import

The above exception was the direct cause of the following exception:

SystemError: <built-in method __contains__ of dict object at 0x000001E4E10C1908> returned a result with an error set

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "scenario_runner.py", line 35, in <module>
    from srunner.scenarioconfigs.openscenario_configuration import OpenScenarioConfiguration
  File "D:\scenario_runner\srunner\scenarioconfigs\openscenario_configuration.py", line 24, in <module>
    from srunner.tools.openscenario_parser import OpenScenarioParser, ParameterRef
  File "D:\scenario_runner\srunner\tools\openscenario_parser.py", line 26, in <module>
    from srunner.scenariomanager.scenarioatomics.atomic_behaviors import (TrafficLightStateSetter,
  File "D:\scenario_runner\srunner\scenariomanager\scenarioatomics\atomic_behaviors.py", line 33, in <module>
    from agents.navigation.basic_agent import BasicAgent
  File "C:\Users\I010503\AppData\Local\Programs\Python\Python37\lib\site-packages\agents\__init__.py", line 21, in <module>
    from . import algorithms
  File "C:\Users\I010503\AppData\Local\Programs\Python\Python37\lib\site-packages\agents\algorithms\__init__.py", line 21, in <module>
    from .ppo import PPO
  File "C:\Users\I010503\AppData\Local\Programs\Python\Python37\lib\site-packages\agents\algorithms\ppo\__init__.py", line 21, in <module>
    from .ppo import PPO
  File "C:\Users\I010503\AppData\Local\Programs\Python\Python37\lib\site-packages\agents\algorithms\ppo\ppo.py", line 27, in <module>
    import tensorflow as tf
  File "C:\Users\I010503\AppData\Local\Programs\Python\Python37\lib\site-packages\tensorflow\__init__.py", line 37, in <module>
    from tensorflow.python.tools import module_util as _module_util
  File "C:\Users\I010503\AppData\Local\Programs\Python\Python37\lib\site-packages\tensorflow\python\__init__.py", line 37, in <module>
    from tensorflow.python.eager import context
  File "C:\Users\I010503\AppData\Local\Programs\Python\Python37\lib\site-packages\tensorflow\python\eager\context.py", line 34, in <module>
    from tensorflow.python.client import pywrap_tf_session
  File "C:\Users\I010503\AppData\Local\Programs\Python\Python37\lib\site-packages\tensorflow\python\client\pywrap_tf_session.py", line 19, in <module>
    from tensorflow.python.client._pywrap_tf_session import *
ImportError: initialization failed

Would someone please help me? 

Thanks a lot

glopezdiest, 2023-03-31:
This isn't really an Scenario Runner issue, as the issue seems to be related with `device_lib` from `tensorflow.python.client`

closed 2023-03-31 by glopezdiest

GhaithNasleh, 2023-04-19:
Thank you for the answer. 
The problem Scenario Runner does not find Agent folder. 

I just copied Agent folder from Carla into Scenario Runner and it works. 

So the methods can be accessed without error feedback.