carla-simulator/scenario_runner #976: Program crashes when encountering curves or intersections during turn with waypoints

jiheddachraoui, 2023-03-27:
I would like to report an issue related to my experience with the openScenario feature in Carla. I have observed that when attempting to make a turn using waypoints, the program operates normally when the waypoints are on a straight line, but crashes when it reaches the first curve or intersection. Despite trying various troubleshooting techniques, I have been unable to resolve the issue and would appreciate assistance in identifying a solution.

glopezdiest, 2023-03-31:
Please, provide more details about your tests
- Versions of CARLA and Scenario Runner
- What commands are you using?
- Any file you created?

jiheddachraoui, 2023-03-31:
hello, i work with version 0.9.13 for both:
> 
<?xml version="1.0" ?>
<OpenSCENARIO>
    <FileHeader revMajor="1" revMinor="0" date="2023-03-23T15:06:44" description="Generated OpenSCENARIO File" author="QGIS OSCGenerator Plugin"/>
    <ParameterDeclarations/>
    <CatalogLocations/>
    <RoadNetwork>
        <LogicFile filepath="Town02"/>
        <SceneGraphFile filepath=""/>
    </RoadNetwork>
    <Entities>
        <ScenarioObject name="Ego_1">
            <Vehicle name="vehicle.ford.mustang" vehicleCategory="car">
                <ParameterDeclarations/>
                <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
                <BoundingBox>
                    <Center x="1.5" y="0.0" z="0.9"/>
                    <Dimensions width="2.1" length="4.5" height="1.8"/>
                </BoundingBox>
                <Axles>
                    <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
                    <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
                </Axles>
                <Properties>
                    <Property name="type" value="ego_vehicle"/>
                </Properties>
            </Vehicle>
        </ScenarioObject>
        <ScenarioObject name="Vehicle_1">
            <Vehicle name="vehicle.jeep.wrangler_rubicon" vehicleCategory="car">
                <ParameterDeclarations/>
                <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
                <BoundingBox>
                    <Center x="1.5" y="0.0" z="0.9"/>
                    <Dimensions width="2.1" length="4.5" height="1.8"/>
                </BoundingBox>
                <Axles>
                    <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
                    <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
                </Axles>
                <Properties>
                    <Property name="type" value="simulation"/>
                </Properties>
            </Vehicle>
        </ScenarioObject>
        <ScenarioObject name="Vehicle_2">
            <Vehicle name="vehicle.ford.ambulance" vehicleCategory="car">
                <ParameterDeclarations/>
                <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
                <BoundingBox>
                    <Center x="1.5" y="0.0" z="0.9"/>
                    <Dimensions width="2.1" length="4.5" height="1.8"/>
                </BoundingBox>
                <Axles>
                    <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
                    <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
                </Axles>
                <Properties>
                    <Property name="type" value="simulation"/>
                </Properties>
            </Vehicle>
        </ScenarioObject>
    </Entities>
    <Storyboard>
        <Init>
            <Actions>
                <GlobalAction>
                    <EnvironmentAction>
                        <Environment name="Environment1">
[... 386 lines cut ...]
            </ConditionGroup>
            <ConditionGroup>
                <Condition name="criteria_RunningRedLightTest" delay="0.0" conditionEdge="rising">
                    <ByValueCondition>
                        <ParameterCondition parameterRef="" value="0.0" rule="lessThan"/>
                    </ByValueCondition>
                </Condition>
            </ConditionGroup>
            <ConditionGroup>
                <Condition name="criteria_RunningStopTest" delay="0.0" conditionEdge="rising">
                    <ByValueCondition>
                        <ParameterCondition parameterRef="" value="0.0" rule="lessThan"/>
                    </ByValueCondition>
                </Condition>
            </ConditionGroup>
            <ConditionGroup>
                <Condition name="criteria_WrongLaneTest" delay="0.0" conditionEdge="rising">
                    <ByValueCondition>
                        <ParameterCondition parameterRef="" value="0.0" rule="lessThan"/>
                    </ByValueCondition>
                </Condition>
            </ConditionGroup>
        </StopTrigger>
    </Storyboard>
</OpenSCENARIO>