carla-simulator/scenario_runner #994: Scenario Configuration error in scenario runner

SExpert12, 2023-06-20:
**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - CARLA Version [e.g. CARLA 0.9.14]
 - Python version [e.g. 3.7]
I refer to [#919](https://github.com/carla-simulator/scenario_runner/issues/919)  but problem is not solved.
 
hi everyone,

I got the same error:
python3 scenario_runner.py --scenario follow_leading_vehicle.py --reloadWorld

warnings.warn(msg, UserWarning)
Configuration for scenario follow_leading_vehicle.py cannot be found!
No more scenarios .... Exiting

I set my path variables in .bashrc file as:
export CARLA_ROOT=/home/sona/Downloads/CARLA14
export SCENARIO_RUNNER_ROOT=/home/sona/Downloads/CARLA14/PythonAPI/carla/scenario_runner
export PYTHONPATH=/usr/bin/python3:/home/sona/Downloads/CARLA14/PythonAPI/carla/dist/carla-0.9.14.-py3.7-linux-x86_64.egg
export PYTHONPATH=/usr/bin/python3:/home/sona/Downloads/CARLA14/PythonAPI/carla/agents
export PYTHONPATH=/usr/bin/python3:/home/sona/Downloads/CARLA14/PythonAPI/carla
export PYTHONPATH=/usr/bin/python3:/home/sona/Downloads/CARLA14/PythonAPI

any suggestion to solve this error

Anhar2, 2023-06-21:
You should not pass the name of the python script, rather the corresponding name of the xml scenario itself found in the examples folder 
without .xml extention, followed by _ then a number representing each scenario number
for instance: FollowLeadingVehicle_1

SExpert12, 2023-06-22:
Yes. It is done now. Do you any idea about changing intersection in CARLA e.g. CARLA provides 4 lane road in 4-way intersection and I need 6-lane road at 4-way intersection. Please help me if you have any idea regarding this.

Thanks for your reply.

Anhar2, 2023-06-22:
Unfortunately I don't know, I suggest you open a new issue so more people can see it

Best of luck

SExpert12, 2023-06-22:
Thanks. I already opened issues but no response from any one. That's why I asked you.

Thanks for your reply.