carla-simulator/scenario_runner #996: Scenario_runner.py is not working windows 10

swapy1, 2023-06-27:
I have installed carla from source and scenario runner also.
carla is working i also tried to play it from UEeditor app. after pressing Play button i was trying to run scenario using scenario runner and i got below error.

**Expected behavior**
C:\scenario_runner>python scenario_runner.py
Traceback (most recent call last):
  File "scenario_runner.py", line 35, in <module>
    from srunner.scenarioconfigs.openscenario_configuration import OpenScenarioConfiguration
  File "C:\scenario_runner\srunner\scenarioconfigs\openscenario_configuration.py", line 24, in <module>
    from srunner.tools.openscenario_parser import OpenScenarioParser, ParameterRef
  File "C:\scenario_runner\srunner\tools\openscenario_parser.py", line 26, in <module>
    from srunner.scenariomanager.scenarioatomics.atomic_behaviors import (TrafficLightStateSetter,
  File "C:\scenario_runner\srunner\scenariomanager\scenarioatomics\atomic_behaviors.py", line 30, in <module>
    import networkx
  File "C:\Users\sgp98\AppData\Roaming\Python\Python38\site-packages\networkx\__init__.py", line 117, in <module>
    import networkx.readwrite
  File "C:\Users\sgp98\AppData\Roaming\Python\Python38\site-packages\networkx\readwrite\__init__.py", line 15, in <module>
    from networkx.readwrite.graphml import *
  File "C:\Users\sgp98\AppData\Roaming\Python\Python38\site-packages\networkx\readwrite\graphml.py", line 314, in <module>
    class GraphML(object):
  File "C:\Users\sgp98\AppData\Roaming\Python\Python38\site-packages\networkx\readwrite\graphml.py", line 346, in GraphML
    (np.int, "int"), (np.int8, "int"),
  File "C:\Users\sgp98\AppData\Roaming\Python\Python38\site-packages\numpy\__init__.py", line 305, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'int'.
`np.int` was a deprecated alias for the builtin `int`. To avoid this error in existing code, use `int` by itself. Doing this will not modify any behavior and is safe. When replacing `np.int`, you may wish to use e.g. `np.int64` or `np.int32` to specify the precision. If you wish to review your current use, check the release note link for additional information.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations

can someone please help me regarding the issue?

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - CARLA Version CARLA 0.9.13
 - Scenario runner version 0.9.13
 - Python version 3.8
 -

closed 2023-06-29 by swapy1

Ep11phany, 2023-07-20:
Hello there, how did you solve this problem? I encountered this one too.

silakuzu, 2025-03-19:
Hi @Ep11phany , you probably solved it now :D but I still wanted to add my solution for the issue for others. This solved the problem:
pip install --upgrade networkx