carla-simulator/scenario_runner #1002: Vehicle not moving in scenarios

ronyshaji, 2023-07-21:
The vehicle in Carla is not moving while running the scenario runner. The pygame windows is opening and I can see the spawned car, but the scenario is not happening. The car is stationary and does not perform the scenario. Other parameters like weather are changing based on the scenario.

**Expected behavior**
The vehicle should perform the scenario as described in the scenario .xml file

**Desktop (please complete the following information):**
 - OS: Windows 10
 - CARLA Version CARLA 0.9.12
 - Python version 3.8
 -

siddharthjain38, 2023-07-21:
@ronyshaji Which scenario you are talking about? Are you talking about ego vehicle or other traffic participant?

ronyshaji, 2023-07-24:
@siddharthjain38  I am taking about the ego_vehicle. I start the scenario_runner.py and then manual_control.py....The pygame windows opens and the vehicle is spawned. but the vehicle is stationary and not moving. In the command line, the sceneraio is still running and completing after some time. but no movement in the pygame.

siddharthjain38, 2023-07-24:
@ronyshaji I tried this scenario and I am able to drive ego vehicle with the help of keyboard. I hope you are controlling ego vehicle and driving it as it is not in autopilot mode.

ronyshaji, 2023-07-25:
@siddharthjain38 So in this scenario also, do we need to run the vehilce manually using the keys? Which scenario does the auto pilot works?

siddharthjain38, 2023-07-25:
@ronyshaji Yes, we have to control ego vehicle manually using keyboard. I don't think they provided any example with autopilot but you can try setting autopilot attribute for ego vehicle.

ronyshaji, 2023-07-26:
@siddharthjain38 Understood. I guess we have to start the vehicle manually but it will do the braking automatically based on scenario. for eg: Pedestrian crossing?

GabrielZZZ, 2023-08-05:
The manual_control.py has an autopilot mode. You can trigger it by pressing 'p' on your keyboard.
[image]

ronyshaji, 2023-08-07:
@GabrielZZZ and @siddharthjain38 Now I understood the working of the scenario. May be I have another question linked to it. How to create a scenario using waypoints? I mean does the waypoints should be equidistant and is it possible to cross the lanes to emulate the lane change?