carla-simulator/scenario_runner #1003: How to create a scenario to park the car in Town 05?

ronyshaji, 2023-07-25:
**Describe our question or idea**
Hello
I am planning to create a scenario in which the car needs to enter the parking lot and park by itself. But while going through the current available scenarios, I was not able to find the solution. Since the waypoint does not goes through the parking lot, I find it difficult. Do you have any idea about how can I do this?

**Expected behavior**
Vehicle enters the parking lot
**Screenshots**
[image]

ronyshaji, 2023-07-25:
@glopezdiest Do you have any feedbacks on this?

ronyshaji, 2023-07-26:
@siddharthjain38 The issue is that there are no waypoints avaialble inside the parking lot. I think its because there is no road associated with it. Do you know any way to create a simple openscenario file to use with carla?

starrye, 2023-10-13:
Try using world postion and then changing carla's get_waypoint method to position it to a custom location