carla-simulator/scenario_runner #1007: Why are there multiple routes? They are interfering with my anticipated route.

shh1v, 2023-07-27:
**Describe our question or idea**
I created a custom routes (_xml_) file that includes a single route which I want the ego vehicle to follow. However, with the debug flag, I learned that there are multiple routes (for e.g. are highway exits) also deployed in the simulation. Initially, the vehicle follows the path from my _xml_ file; but, when it approaches one of the exits, it takes the exit route. I am not sure why the other routes are being deployed by default. I do not want that.

**Expected behavior**
The ego vehicle should exclusively follow my route specified in the _xml_ file.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - CARLA Version 0.9.13
 - Python version 3.7
 - Version 0.9.13

shh1v, 2023-07-27:
Here is a more accurate image describing the unwanted behavior.

[image]

shh1v, 2023-07-28:
Any thoughts @glopezdiest? I would really appreciate them.

shh1v, 2023-08-02:
I noticed that this behavior is dependent on the route taken. If the route overlaps one of the junction turn routes or is close enough, that route is considered in the scenario. How do I disable this behavior?

shh1v, 2023-08-03:
Hey, Could you please help me out @fabianoboril. I have been stuck here for quite a while.

shh1v, 2023-08-04:
Okay, so I have figured out a workaround; however, the bugs still remain. I will have to give context first; I am using DReyeVR vehicle for a research driving simulator study, which has integrated Logitech steering wheels and pedals. This allowed me to very accurately drive and record the waypoints for my route in the world. Now, I was logging all the points very densely, i.e., each adjacent waypoint possible. And, so I think `downsample_route `came into play, or the `route_trace `was not working as expected? When I recorded a less dense route (30 meters apart), the route following was working flowless for the same planned route.

I am keeping this issue open, as the bug still remains. I would love to hear an actual solution to this!

ronyshaji, 2023-08-07:
@shh1v Hey, As you mentioned about the waypoint collection, does this collect the waypoint of the locations whereever the vehicle is driven ? I mean i have to create a scenario with waypoints and it will nice to have a method to collect the waypoints.

shh1v, 2023-08-07:
Hello @ronyshaji, If you are utilizing [DReyeVR](https://github.com/HARPLab/DReyeVR) for your project, you'll find it quite convenient to create the necessary routes. However, to assist you further, I'm providing the [code](https://github.com/shh1v/scenario_runner/tree/master/routegenerator) I personally used. This code will not only help you generate the route but also display it within the CARLA world (for debugging), and subsequently write it into an XML file.

ronyshaji, 2023-08-08:
Hey @shh1v Thanks for providing the code. What I am trying to do is that to park the vehicle in one of the parking lot in Town 04. But the opendrive doesnot have any waypoints going through that. So is it possible to record the parking lot waypoints also ? I will try the code and check. Thanks !!

shh1v, 2023-08-08:
There exists no waypoints in the parking lot. Even if you record location(x, y, z) points using a script (e.g., the one I provided), the internal mechanism of `scenario_runner `converts the points you provided into the closest available waypoint. This is because it stores a graph that has all the waypoints as nodes, and edges that represent `RoadOption `(`Follow`, `ChangeLane`, etc.), i.e, what the actor has to do to reach the nodes. When you provide waypoints, it searches through the graph and finds a route for you. Thus, you cannot use _any_ recorded `carla.Location(x, y, z)` points to create a route (as this is not even possible).

The only solution, which I think, after going through their source code, is to extend this internal graph by adding waypoints having location and rotation present in the parking lot. And, this would be a challenge.

Lastly, the script I provided was catered for DReyeVR, and was just for reference. However, if you want to use it, you will have to run `manual_control.py` and modify my script to get a reference to the ego vehicle. Then drive around and it will collect the waypoint. Also, you will have to write methods to change the settings to asynchronous, as I have not provided code for the helper methods. It would not be a challenge. Reference the Python API documentation. Let me know if you need further help with this.

ronyshaji, 2023-08-08:
Hey @shh1v 
So as far as I understood, we can only run vehicles through the waypoints from opendrive while creating scenario with waypoint. But once i created a scenario with random waypoints which is going through the sidewalk and while running it traces the path as specifed by waypoints.

Also based on your comments, then how does pedestrain crossing works? I think it is also based on waypoints which are not available on the map.

shh1v, 2023-08-08:
@ronyshaji Right, it is able to plan and execute routes having waypoints situation on the road only. As far as I can understand, in the `global_route_planner.py`, it gets the waypoints _on the road_ [through CARLA server](https://github.com/carla-simulator/carla/blob/43b5e7064872bb6a9529664c2218e29df38dca04/PythonAPI/carla/agents/navigation/global_route_planner.py#L84), and then [builds a graph](https://github.com/carla-simulator/carla/blob/43b5e7064872bb6a9529664c2218e29df38dca04/PythonAPI/carla/agents/navigation/global_route_planner.py#L124C15-L124C15). Thus, it only supports routes that are on the road.

Looking at the documentation, it also has [defined set of waypoints](https://carla.readthedocs.io/en/latest/tuto_M_generate_pedestrian_navigation/#pedestrian-navigable-meshes) for pedestrians that they can navigate through. They can not navigate anywhere in the world, unless a [custom cross-walk is created](https://carla.readthedocs.io/en/latest/tuto_M_generate_pedestrian_navigation/#generate-the-pedestrian-navigation).

shh1v, 2023-08-09:
Hello @ronyshaji.

I am not sure. I am not working with OpenScenario, but rather python scenario modules of `scenario_runner` as it is quite easy to pick up. But I can tell you that it is possible using the [python scenario classes](https://carla-scenariorunner.readthedocs.io/en/latest/creating_new_scenario/), as negative `target_speed` is supported by the [`LongitudnalVehicleControl`](https://github.com/carla-simulator/scenario_runner/blob/f1a94516db57612f22d5c42e597e350e781b9224/srunner/scenariomanager/actorcontrols/vehicle_longitudinal_control.py#L19) controller class. Furthermore, I think [`WaypointFollower`](https://github.com/carla-simulator/scenario_runner/blob/f1a94516db57612f22d5c42e597e350e781b9224/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py#L1872) should also work, but I have not tested it. You will have to find your way in OpenScenario. I am also just starting out with `scenario_runner`. Sorry.

ronyshaji, 2023-08-09:
@shh1v Thanks for the feedbacks. I tried with waypoints but the movement of vehicles is not that much realistic. Also the wheels are not tilted while using waypoints, rather teh whole vehicle just slides to next points. Hopefully I can find something more in openscenario soon.