carla-simulator/scenario_runner #1011: OpenScenario is not working with waypoints based scenario

ronyshaji, 2023-08-07:
Hey all

I have successfully generated an OpenScenario file using qgis, where I have specified an ego vehicle and configured it to adhere to a series of waypoints. These waypoints were defined within the 'Town01' map on qgis. However, upon attempting to execute the .xosc file, I encountered an issue where the vehicle simply proceeded in a straight line rather than following the designated waypoints.

For your reference, I have provided a link to the OpenScenario file - [WaypointScenario](https://drive.google.com/file/d/1DxVdg099qn63o38w-7Ri20MWzz9Mjri9/view?usp=drive_link).

I have explored potential solutions by creating varied waypoints across different sections of the map, yet the problem persists. I am uncertain whether the waypoints need to be evenly spaced, and I am also curious about handling lane changes within the scenario.

Thank you for your guidance in resolving this matter.

shh1v, 2023-08-07:
This is somewhat what the erratic behavior I observed too. Looking at your file, it seems that the waypoints you recorded are very close? I am uncertain if this is what's causing the issue, but for me, recording waypoints very wide apart (e.g., 30 meters) solved the issue. I have talked about this [here](https://github.com/carla-simulator/scenario_runner/issues/1007).

ronyshaji, 2023-08-08:
@shh1v My use case is limited for only 5 meters with the vehicle entering into a parking lot. Thats why I choosed the close waypoints. But the vehicle is not even tracing the waypoints mentioned. It is spawning at the spawn point from the openscenario file and moves forward straight. That is confusing for me. Do we need to provide the value of 'h'(yaw) also to make the vehicle take a turn?

shh1v, 2023-08-08:
Please see my comment [here](https://github.com/carla-simulator/scenario_runner/issues/1007).

PrussianBbblue, 2023-08-24:
Hi, I have located a bug of CARLA and it might the cause of your problem. 
Firsly, Your description seems more appropriate to use "followTrajectoryAction". Acording to the definition of openscenario, the vehicle follows a specific points of a trajectory using this action.
However, the vehicle always driving in the middle of the lane rather than follow your trajectory even if you use "followTrajectoryAction". It is a bug of Carla PythonAPI "carla.Map.get_waypoint()". The parameter "project_to_road" does not work at all, which means the method always return a point at the center of the lane no matter you use True or False for this para. So what really deliveried to the Carla controller is the points at the center of the lane instead of your pre defined points.
It is hard problem. I tried to rewrite some python files in scenario_runner and Carla PythonAPI to fixed it.

ronyshaji, 2023-08-25:
@PrussianBbblue So you mean we can able to drive the vehicle whereever we wanted other than through waypoints if i used the followTrajectoryAction ?

PrussianBbblue, 2023-08-25:
@ronyshaji Theoretically, yes. But it still works as AssignRouteAction due to a bug of CARLA PythonAPI as I mentiond above.

ronyshaji, 2023-08-25:
@But in tht case, i can create a scenario in a parking lot in Town04 like parking our vehicle using any of the methods ?  unfortunately there is no waypoints availáble in the parking lot.

PrussianBbblue, 2023-08-25:
@ronyshaji If there is no bug, it should works as you wish using followTrajectoryAction.
I rewrite some files in scenario runner and Carla to use carla.Transform rather than carla.Waypoint to control the movement of the vehicle so that the vehicles don't have to stay at the center of the lane. And it does work well. You may have a try if neccesarry.

ronyshaji, 2023-08-25:
Hey can you share the link where you have the modified file? Also, do we need to create the openscenario with waypoints here or any other method?

PrussianBbblue, 2023-08-28:
@ronyshaji Hi, sorry I cannot provide the code that I modified cuz it's part of work of the company. But I can say the key is to use transform rather than waypoint in npc_vehicle_control(in scenario_runner), local_planner and controller (in Carla).
And yes, you can manage your car drive anywhere on the road using followTrajectoryAction.

ronyshaji, 2023-08-28:
@PrussianBbblue I think its like a full editing of the whole code as i see that it uses the waypoint in many areas. and it would be great if you can share your discord id or something so that i can ask you directly rather than here. Thanks for the feedback.

ronyshaji, 2023-08-31:
@PrussianBbblue Did you ever used the **<StoryboardElementStateCondition storyboardElementType="action" storyboardElementRef="MovingForward" state="completeState"/>** in scenario runner? It is not working for me and due to this the next event is not happening. Did you ever tried this while using the scenario runner?

18303898905, 2024-08-30:
@ronyshaji Can you run this .xosc file by using scenario runner successfully ? I can not make it.