carla-simulator/scenario_runner #1013: Running scenario runner on windows

WheelWil, 2023-08-11:
I am trying to verify if scenario runner runs by "python scenario_runner.py --help" or "python scenario_runner.py --vers", I am getting this:

(carlaSim) C:\ws\913\WindowsNoEditor\scenario_runner>python scenario_runner.py --vers
Traceback (most recent call last):
  File "scenario_runner.py", line 37, in <module>
    from srunner.scenarioconfigs.openscenario_configuration import OpenScenarioConfiguration
  File "C:\ws\913\WindowsNoEditor\scenario_runner\srunner\scenarioconfigs\openscenario_configuration.py", line 24, in <module>
    from srunner.tools.openscenario_parser import OpenScenarioParser, ParameterRef
  File "C:\ws\913\WindowsNoEditor\scenario_runner\srunner\tools\openscenario_parser.py", line 26, in <module>
    from srunner.scenariomanager.scenarioatomics.atomic_behaviors import (TrafficLightStateSetter,
  File "C:\ws\913\WindowsNoEditor\scenario_runner\srunner\scenariomanager\scenarioatomics\atomic_behaviors.py", line 32, in <module>
    from agents.navigation.basic_agent import BasicAgent, LocalPlanner
  File "C:\ws\913\WindowsNoEditor\PythonAPI\carla\agents\navigation\basic_agent.py", line 14, in <module>
    from shapely.geometry import Polygon
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\shapely\geometry\__init__.py", line 4, in <module>
    from .base import CAP_STYLE, JOIN_STYLE
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\shapely\geometry\base.py", line 19, in <module>
    from shapely.coords import CoordinateSequence
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\shapely\coords.py", line 8, in <module>
    from shapely.geos import lgeos
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\shapely\geos.py", line 154, in <module>
    _lgeos = CDLL(os.path.join(sys.prefix, 'Library', 'bin', 'geos_c.dll'))
  File "C:\Users\user\.conda\envs\carlaSim\lib\ctypes\__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] The specified module could not be found

 - OS: Win10
 - CARLA Version 0.9.13 (binary)
 - Python version 3.7 + env variables (conda)
 - Scenario Runner Version 0.9.13

ravivaghasiya1998, 2023-08-31:
pip uninstall shapely
pip install shapely

try to uninstall shapely and reinstall it. This solution worked for me

WheelWil, 2023-08-31:
Thank you for the solution! :)

Turuu1124, 2024-08-16:
thank you, been stuck trying to make it work for days now