carla-simulator/scenario_runner #1016: How we can create time based trigger in route scenario rather than location based

siddharthjain38, 2023-08-14:
**Describe our question or idea**
How we can create time based trigger in route scenario rather than location based. 

**Expected behavior**
I wants to create time based trigger means execute any trigger only after specific amount of time rather than location based. We do not want to trigger scenario though ego vehicle came to trigger point before specific or given amount of time. 

For example initially we can drive ego vehicle drive for 5-10 before scenario starts triggering in route scenario.

shh1v, 2023-08-14:
Why use scenario runner at all, then? You can simply build what you want with the Python client API, no?

siddharthjain38, 2023-08-16:
@shh1v I wants to implement sequence of scenario like cut-in, close following and We have to run these scenario in single route.