carla-simulator/scenario_runner #1020: Teleport issue in OpenScenario

ronyshaji, 2023-08-25:
Hello
I am working with OpenScenario and in my case the teleportation to the specified point in the scenario file is not happening. For precise, I am using ros scenario runner and firstly the vehicle is spawned at a random location. Then while executing the scenario it should teleport to the specified location.

I was checking the atomic behaviour which does the teleportation and was not able to find anything in the scenario runner. Or is it use any atomic bahaviour to teleport the vehicle. 

From the scenario_runner.py [Line 232](https://github.com/carla-simulator/scenario_runner/blob/f1a94516db57612f22d5c42e597e350e781b9224/scenario_runner.py#L232) the vehicle is setting a new transform based on the transform from the scenario file.

So I have to ask where is exactly the teleportation happening?

**Expected behavior**
Teleport to the specified location from OpenScenario file.

**Desktop (please complete the following information):**
 - OS: Win 10
 - CARLA Version : CARLA 0.9.12
 - Python version : 3.8

PrussianBbblue, 2023-08-25:
Do you use TeleportAction in Storyboard Init? It's the section where you use actions to specify the inital state of objects (speed, position etc.).
If you don't, your vehicles would be located at (0, 0, 0).

ronyshaji, 2023-08-25:
@PrussianBbblue Yes. It is available in the Openscenario file. But my vehicle model is a simulink vehicle. So not sure what is the issue is.

closed 2023-11-07 by ronyshaji